fsphil/hacktv #2: Bring back Firefly

ikarus23, 2017-05-24:
I know the readme says it won't. But please reconsider this! I really need this feature!
I totally trust in in your coding skills! You can do it! And if not, maybe someone in the
community is able to issue a pull request to resolve this?!

fsphil, 2017-05-24:
Only versions 1.01 to 1.14 are supported. Waiting on further updates from upstream who have been unresponsive after the project was forked.

veso266, 2017-08-17:
can anyone please tell me what Bring back Firefly is? (the onylthing I was able to find was a television show: http://io9.gizmodo.com/theres-still-a-way-firefly-could-come-back-to-televisi-1521723457 )

Thanks for Anwsering
PS: do you think you could compile hacktv for windows (since libhackrf exist for windows and GNURadio works on windows nicly too: https://github.com/pothosware/PothosSDR/wiki/GNURadio)

Thanks for Anwsering and Best Regards

fsphil, 2017-08-23:
Firefly is a long term goal :-)

I don't have Windows here but I don't see why it's not possible. hacktv does not use gnuradio, but it will need libhackrf and ffmpeg to compile.

Teletext will probably happen at some point, yes.

veso266, 2017-08-24:
Thanks so what is Firefly? (Is this analog-tv encryption method like Nagravision and Videocrypt (will those be added at some point?)

Thanks for Anwsering and Best Regards

fsphil, 2017-08-24:
Yes it's the TV show :) Although I am now tempted to make an encryption system named Firefly.

I have just added Videocrypt support, see commit 276d76521cb8f2a2f20b9417767cf5b6cb7d1aa0.
I'm currently researching Nagravision.

veso266, 2017-08-24:
yea Thanks now I just need to find a Videocrypt Descrambler (is there any software descrambler?)(can PAL L television handle it?)

you think you could add Zweikanaltone (A2 Stereo): https://en.wikipedia.org/wiki/Zweikanalton
PS: do you think you could add clock and date and name of the station to Philips PM5544 Test Card? (something like this: https://www.youtube.com/watch?v=KUenRyt5u5Y)

Thanks for Anwsering and Best Regards
PS: if you need it there is also python implementation for NTSC: https://github.com/argilo/sdr-examples/blob/master/ntsc/ntsc-encode.py

fsphil, 2017-08-24:
It can already transmit NTSC.
PAL-L video is identical to PAL-I, it would be trivial to add.

You need to open separate issues for these :)

veso266, 2017-08-24:
OK Openned separate issue for zweikanaltone and Test Card: https://github.com/fsphil/hacktv/issues/3

kipz, 2020-12-04:
Any update on this please?

veso266, 2020-12-26:
Hi there, sorry for late replay (don't have much time for playing right now :smile: )

after compiling the new version (had to change theese 2 in mac.c (line 451)
```c
	/* Get the timezone offset */
	#ifdef __MINGW32__
		localtime_s(&timestamp, &tm);
		i = tm.tm_mon / 1800;
	#else
		localtime_r(&timestamp, &tm);
		i = tm.tm_gmtoff / 1800;
	#endif
	if(i < 0) i = -i | (1 << 5);
	
	/* Calculate Modified Julian Date */
	#ifdef __MINGW32__
		gmtime_s(&timestamp, &tm);
	#else
		gmtime_r(&timestamp, &tm);
	#endif
```

I tried it on PAL-L (for some reason I had to set picture to SECAM not PAL, but sound could be in PAL-L)

the colors were a bit wierd (but it could only be my television set that does wierd things
https://youtu.be/2jO025ZIPKc

sorry for the audio (some morons decided that the music that was playing is copyrighted (even they have no idea what I am doing :smile:)

fsphil, 2021-01-09:
hacktv's `l` mode is always secam. If you want to add a pal-l mode you'll need to edit video.c.

Please open a new issue for these types of requests, unless it's related to the TV show Firefly :-)