fsphil/hacktv #3: Suggestions

veso266, 2017-08-24:
you think you could add Zweikanaltone (A2 Stereo): https://en.wikipedia.org/wiki/Zweikanalton
PS: do you think you could add clock and date and name of the station to Philips PM5544 Test Card? (something like this: https://www.youtube.com/watch?v=KUenRyt5u5Y)

Thanks for Anwsering and Best Regards

fsphil, 2017-08-25:
It seems that none of my TVs support this mode of stereo. I won't be able to do any testing.

DG9BFC, 2018-03-21:
hmmm even if your tv set can not use that stereo decoding i have an idea for you ..... use an (analog) sat receiver as a test receiver (can be found used in evilbay for a few bucks)
they can use that stereo signal with 5.5/5.74 megs carrier (or 7.02/7.20)
some use 180 kc difference and some use 240kc (selectablde in sat receiver menue)
best would be two selectable carriers (say 5.5 5.74) ... and switchable of you wanne send
L / R .... or L+R / R ... or L+R / L-R ... and if you wanna add a pilot tone and what frequency the pilot tone should have ... then you have a complete flexible solution for any stereo system with low interaction to the picture signal
if you also can send 6 and 6.5 megs tone carrier on 2.4 gig then a "normal" videolink box can be used as a receiver (those tiny videolink boxes to have the sat receiver in the living room while a second tv is in the sleeping room)

ljones0, 2019-09-04:
Hello,

I can get pretty much most things to work with hackrf which is good news. But for 405 line is it possible to add audio to the video at all? 405 line transmissions for video seem to be working pretty well btw.

ljones

fsphil, 2019-09-04:
System A should already have AM audio at -3.5 MHz. I recently got a Sony 9-90UB for 405-line testing and I can hear the audio but it's very quiet, which I'm fairly certain is a fault with the Sony.

ljones0, 2019-09-04:
:-) Snap! I have the exact same TV so I guess my problem is the same (I also got it to try hacktv/hackrf).

Just off the top of my head would be of any use (maybe in the examples) to list which different computers work (and which don't?). 2 PCs (ryzen5 and hp stream 11) worked but a pi did not; maybe keep a list?

ljones

fsphil, 2019-09-04:
Can you try tweaking these lines in video.c: https://github.com/fsphil/hacktv/blob/fec728d4c123156fbadcbd2f54f1959ff4d1cf64/video.c#L504-L505

I've seen some hardware 405 modulators that use the same level for both audio and video carriers, which would be 0.5 and 0.5 here. I have already tried this here and although it does improve the audio volume it also distorts the picture. But this could be a fault with my Sony, you may have better luck.

It would be difficult maintaining a list of what computers work, it depends on too many factors. The Pi's main bottleneck is USB bandwidth, if you try reducing the sample rate you might find it works better. Eg.:

`# hacktv -s 8000000 -f 66750000 -m a -g 47 test`

ljones0, 2019-09-08:
Ok I gave that a go. I could certianly hear something if I used 0.5 and 0.5 but the audio this time was marred by buzz from the video. Although I will confess my 9-90ub is *far* from being 100% (it needs a fair bit of work). But looks like it is doing something. I did try also reducing the rate on the pi which worked reasonably well although I did have some strange colour pattens on the picture.

ljones

fsphil, 2019-09-08:
It sounds like our TVs might have similar issues, it would be nice to try it on a different model. I'm trying _very_ hard not to use this as an excuse to get one, I've run out of space :-)

PAL probably won't work at 8 MHz, the colour and audio sub-carriers would end up in the wrong place. 405 should be fine though.

closed 2019-09-21 by fsphil