fsphil/hacktv #4: can't compile it for windows on linux

veso266, 2017-08-27:
Hi after getting hacktv working on linux I decided to try and compile it for windows since I have a hard time compiling c on windows I decided to try to compile hacktv for windows on linux using mingw cross-compiler

I followed this tutorial: https://arrayfire.com/cross-compile-to-windows-from-linux/

and changed makefile a from to this

```make
CC=gcc
CFLAGS=-g -Wall -pthread -O3
LDFLAGS=-g -lm -pthread

CFLAGS+=`pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil`
LDFLAGS+=`pkg-config --libs libavcodec libavformat libswscale libswresample libavutil`

CFLAGS+=`pkg-config --cflags libhackrf`
LDFLAGS+=`pkg-config --libs libhackrf`

OBJS=hacktv.o video.o videocrypt.o nicam728.o test.o ffmpeg.o hackrf.o

all: hacktv

hacktv: $(OBJS)
	$(CC) -o hacktv $(OBJS) $(LDFLAGS)

.c.o:
	$(CC) $(CFLAGS) -c $< -o $@

install:
	cp -f hacktv /usr/local/bin/

clean:
	rm -f *.o hacktv

```
to this:
```make
CC=x86_64-w64-mingw32-gcc
CFLAGS=-g -Wall -pthread -O3
LDFLAGS=-g -lm -pthread

CFLAGS+=`pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil`
LDFLAGS+=`pkg-config --libs libavcodec libavformat libswscale libswresample libavutil`

CFLAGS+=`pkg-config --cflags libhackrf`
LDFLAGS+=`pkg-config --libs libhackrf`

OBJS=hacktv.o video.o videocrypt.o nicam728.o test.o ffmpeg.o hackrf.o

all: hacktv

hacktv: $(OBJS)
	$(CC) -o hacktv $(OBJS) $(LDFLAGS)

.c.o:
	$(CC) $(CFLAGS) -c $< -o $@

install:
	cp -f hacktv /usr/local/bin/

clean:
	rm -f *.o hacktv
```

but when running make I get a weird error (it didn't complain about librarys which I assumed):
 ```bash
mitja@ubuntu:~/analog-tv/hacktv$ make
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `pkg-config --cflags libhackrf` -c hacktv.c -o hacktv.o
In file included from hacktv.c:18:0:
/usr/include/stdio.h:102:19: error: conflicting types for ‘ssize_t’
 typedef __ssize_t ssize_t;
                   ^
In file included from /usr/share/mingw-w64/include/stddef.h:7:0,
                 from /usr/lib/gcc/x86_64-w64-mingw32/4.9-win32/include/stddef.h:1,
                 from /usr/include/stdio.h:33,
                 from hacktv.c:18:
/usr/share/mingw-w64/include/crtdefs.h:45:35: note: previous declaration of ‘ssize_t’ was here
 __MINGW_EXTENSION typedef __int64 ssize_t;
                                   ^
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:132:0,
                 from /usr/include/stdlib.h:314,
                 from hacktv.c:19:
/usr/include/time.h:75:18: error: conflicting types for ‘time_t’
 typedef __time_t time_t;
                  ^
In file included from /usr/share/mingw-w64/include/stddef.h:7:0,
                 from /usr/lib/gcc/x86_64-w64-mingw32/4.9-win32/include/stddef.h:1,
                 from /usr/include/stdio.h:33,
                 from hacktv.c:18:
/usr/share/mingw-w64/include/crtdefs.h:138:20: note: previous declaration of ‘time_t’ was here
 typedef __time64_t time_t;
                    ^
In file included from hacktv.h:21:0,
                 from hacktv.c:23:
/usr/include/stdint.h:122:27: error: conflicting types for ‘uintptr_t’
 typedef unsigned long int uintptr_t;
                           ^
In file included from /usr/share/mingw-w64/include/stddef.h:7:0,
                 from /usr/lib/gcc/x86_64-w64-mingw32/4.9-win32/include/stddef.h:1,
                 from /usr/include/stdio.h:33,
                 from hacktv.c:18:
/usr/share/mingw-w64/include/crtdefs.h:75:44: note: previous declaration of ‘uintptr_t’ was here
 __MINGW_EXTENSION typedef unsigned __int64 uintptr_t;
                                            ^
In file included from hacktv.h:21:0,
                 from hacktv.c:23:
/usr/include/stdint.h:145:0: warning: "__INT64_C" redefined
 #  define __INT64_C(c) c ## L
 ^
<built-in>: note: this is the location of the previous definition
In file included from hacktv.h:21:0,
                 from hacktv.c:23:
/usr/include/stdint.h:146:0: warning: "__UINT64_C" redefined
 #  define __UINT64_C(c) c ## UL
 ^
<built-in>: note: this is the location of the previous definition
Makefile:20: recipe for target 'hacktv.o' failed
make: *** [hacktv.o] Error 1
 ```

I would realy like to compile this for windows because I can't use my HackRF in a VM and its unpractical to boot to linux just to have a play with an awesome project I dream about when I got my HackRF One

Thanks for Anwsering and Best Regards

fsphil, 2017-08-27:
The error you're seeing looks like a gcc or libc issue. I did try compiling it with mingw64 on my machine (Fedora 25 x86_64) and it complains about the libraries as expected.

If I disable ffmpeg and libhackrf it produces hackrf.exe, but it won't be much use to anyone without those two libraries :) I may try building those later.

veso266, 2017-08-29:
did you have any luck compiling libraries?

fsphil, 2017-09-06:
They both compile fine on Fedora 25, but I have no way to test them.
I didn't try building hacktv with them either.

veso266, 2017-09-06:
If you manage to compile hacktv with them I would be happy to test it

PS: do you think you can post steps you took to compile them? (and how to link them to hacktv)

Thanks for Anwsering and Best Regards

fsphil, 2017-09-13:
I followed these guides:

ffmpeg: https://trac.ffmpeg.org/wiki/CompilationGuide/MinGW
libhackrf: https://github.com/mossmann/hackrf/tree/master/host

veso266, 2017-09-14:
Thanks for instructions

after installing Fedora 26 64 Bit into a VM, installing MinGW (and everything that comes with it (I used apper package manager to install everything that had mingw64 in it) and I also installed mingw64-gsl so I got Mingw version of pkg-config

then I compiled this version of libhackrf with MinGW per instructions: https://github.com/mossmann/hackrf/releases/download/v2015.07.2/hackrf-2015.07.2.tar.xz
and found ffmpeg libs for windows online: https://ffmpeg.zeranoe.com/builds/win64/dev/ffmpeg-20170913-ec1573f-win64-dev.zip

then I just setted PKG_CONFIG_PATH variable to my pkg-config files and proceded to compile hacktv
it went well (some warnings in ffmpeg.c) until it hitted hackrf.c part where it started to complain

```bash
[mitja@localhost hacktv]$ make
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c hacktv.c -o hacktv.o
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c video.c -o video.o
video.c: In function 'vid_next_line':
video.c:1046:30: warning: 'lut_b' may be used uninitialized in this function [-Wmaybe-uninitialized]
    s->output[x * 2] += (lut_b[x] * s->burst_level) >> 16;
                              ^
video.c:972:8: warning: 'seq' may be used uninitialized in this function [-Wmaybe-uninitialized]
  if(seq[0] == 'v') w = s->vsync_short_width;
     ~~~^~~
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c videocrypt.c -o videocrypt.o
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c nicam728.c -o nicam728.o
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c test.c -o test.o
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c ffmpeg.c -o ffmpeg.o
ffmpeg.c: In function '_av_ffmpeg_read_video':
ffmpeg.c:142:3: warning: 'avcodec_decode_video2' is deprecated [-Wdeprecated-declarations]
   avcodec_decode_video2(av->video_codec_ctx, av->frame, &i, &packet);
   ^~~~~~~~~~~~~~~~~~~~~
In file included from ffmpeg.c:18:0:
includes/libavcodec/avcodec.h:4971:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^~~~~~~~~~~~~~~~~~~~~
ffmpeg.c: In function '_av_ffmpeg_read_audio':
ffmpeg.c:190:3: warning: 'avcodec_decode_audio4' is deprecated [-Wdeprecated-declarations]
   avcodec_decode_audio4(av->audio_codec_ctx, av->frame, &i, &packet);
   ^~~~~~~~~~~~~~~~~~~~~
In file included from ffmpeg.c:18:0:
includes/libavcodec/avcodec.h:4922:5: note: declared here
 int avcodec_decode_audio4(AVCodecContext *avctx, AVFrame *frame,
     ^~~~~~~~~~~~~~~~~~~~~
ffmpeg.c:205:10: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
    frame = av->frame_s16[0];
          ^
ffmpeg.c: In function 'av_ffmpeg_open':
ffmpeg.c:275:3: warning: 'codec' is deprecated [-Wdeprecated-declarations]
   if(av->video_stream == NULL && av->format_ctx->streams[i]->codec->codec_type == AVMEDIA_TYPE_VIDEO)
   ^~
In file included from ffmpeg.c:19:0:
includes/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
ffmpeg.c:280:3: warning: 'codec' is deprecated [-Wdeprecated-declarations]
   if(av->audio_stream == NULL && av->format_ctx->streams[i]->codec->codec_type == AVMEDIA_TYPE_AUDIO)
   ^~
In file included from ffmpeg.c:19:0:
includes/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
ffmpeg.c:295:2: warning: 'codec' is deprecated [-Wdeprecated-declarations]
  av->video_codec_ctx = av->video_stream->codec;
  ^~
In file included from ffmpeg.c:19:0:
includes/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
ffmpeg.c:325:2: warning: 'avpicture_get_size' is deprecated [-Wdeprecated-declarations]
  len = avpicture_get_size(AV_PIX_FMT_RGB32, s->active_width, s->conf.active_lines);
  ^~~
In file included from ffmpeg.c:18:0:
[... 15 lines cut ...]
     AVCodecContext *codec;
                     ^~~~~
x86_64-w64-mingw32-gcc -g -Wall -pthread -O3 `mingw64-pkg-config --cflags libavcodec libavformat libswscale libswresample libavutil` `mingw64-pkg-config --cflags libhackrf` -c hackrf.c -o hackrf.o
hackrf.c:207:12: error: conflicting types for '_write'
 static int _write(void *private, int16_t *iq_data, size_t samples)
            ^~~~~~
In file included from /usr/x86_64-w64-mingw32/sys-root/mingw/include/unistd.h:10:0,
                 from hackrf.c:23:
/usr/x86_64-w64-mingw32/sys-root/mingw/include/io.h:225:23: note: previous declaration of '_write' was here
   _CRTIMP int __cdecl _write(int _FileHandle,const void *_Buf,unsigned int _MaxCharCount);
                       ^~~~~~
hackrf.c:232:12: error: conflicting types for '_close'
 static int _close(void *private)
            ^~~~~~
In file included from /usr/x86_64-w64-mingw32/sys-root/mingw/include/unistd.h:10:0,
                 from hackrf.c:23:
/usr/x86_64-w64-mingw32/sys-root/mingw/include/io.h:192:23: note: previous declaration of '_close' was here
   _CRTIMP int __cdecl _close(int _FileHandle);
                       ^~~~~~
make: *** [Makefile:20: hackrf.o] Error 1
```

and here is hacktv I tried to compile: http://upload.dxing.si/server/php/files/hacktv.7z

Thanks for Anwsering and Best Regards

veso266, 2018-06-15:
well after learning a bit of c 😄 and a few month later I tried again and it finaly compiled (just had to rename _write and _close to something else (_writeRF and _closeRF) as they were already declared with that name in io.h and unistd.h in MinGW)

also have to change gmtime_r to gmtime_s in teletext.c because windows doesn't have gmtime_r
https://stackoverflow.com/questions/19051762/difference-between-gmtime-r-and-gmtime-s

I feel souch and idiot right now (because I complained about simple stuff) but at least I learned something new 😄

closed 2018-06-15 by veso266