fsphil/hacktv #6: SECAM

SopaXorzTaker, 2018-02-08:
Does this support SECAM?

It would be more interesting if it did, as the standard is quite obscure and I still don't have any idea how the color is modulated...

fsphil, 2018-02-08:
Not yet, but it's on my TODO list. SECAM colour is encoded by frequency modulating the colour sub-carrier with the U and V signals, alternating between them each line.

SopaXorzTaker, 2018-02-08:
I know that, but do you have any detailed documentation?
Or just the subcarrier frequencies and deviations?

fsphil, 2018-02-08:
There is a lot of good information here: http://www.radios-tv.co.uk/Pembers/World-TV-Standards/Colour-Standards.html#SECAM

SopaXorzTaker, 2018-02-10:
How could I miss that?!
Thanks, it's really useful!

Potomac, 2019-04-01:
Hello Fsphil,

Did you make progress about secam implementation ?

You can add it in hackTV as experimental option, not for using directly with hackRF device, but just for generating on hard-disk a secam baseband file from a video file (h264) with ffmpeg, something like : 

$ hacktv -o baseband.bin -m secam example.mkv

For the moment nobody has managed to create a secam software encoder/decoder, it would be great if you can implement secam software algorithm (even if it is not perfect),

thanks again for this wonderful software

fsphil, 2019-04-14:
I did a few tests and was able to get colour, but it was very poor. I could add it, maybe with a warning about how bad it is :)

Potomac, 2019-04-15:
Thanks, it's a good news if you can implement as a experimental feature, 

it would be great also to have 2 secam versions about color identification :

- Field identification: by chrominance subcarrier reference signals occupying nine lines of field-blanking period on Lines 7 to 15 in Fields 1 & 3 and Lines 320 to 328 in Fields 2 & 4, this mode was used by old secam tv set in France between 1967 and 1979
 
- Line identification: by chrominance subcarrier reference signals inserted into the back porch of the line blanking period, this mode was used by modern secam tv in france (1980 to the end of secam broadcast, in 2011)

more information here : 
https://www.radios-tv.co.uk/Pembers/World-TV-Standards/Colour-Standards.html#SECAM

Zcooger, 2019-04-15:
Hello, for your information both color reference signals can be transmitted simultaneously (as it was done during transition period), so there is no need for 2 separate SECAM versions.

Potomac, 2019-04-16:
@Zcooger : yes both methods can be transmitted simultaneously, but it will trigger a problem : the VBI space will be "eat" a little by the field identification method ( called "green bottles", "les bouteilles" in french), which can be a problem if you want to transmit complex teletext, VBI data,

that's why a "line identification" mode can be interesting to solve this problem,

we can create 3 modes : 

- field identification only : useful for old tv sets produced before december 1979, because they display colors only with this color identification method, the other method will give black and white images on these old tv sets

- line identification only : suitable for secam tv sets built after december 1979

- simultaneous method (field + line) : used by french tv channels in the 80's and 90's (except canal+ which uses only line identification method), after 2005 all channels use color  line identification method

fsphil, 2019-06-12:
I've just pushed very basic SECAM support. It will produce colour but it's not very good.

Uses line identification.

Potomac, 2019-06-12:
@fsphil  : It's great, thanks :+1: 

Do you plan to add "field identification" feature for secam ?

fsphil, 2019-06-12:
Do you think it's needed? Are there TVs that require it?

Potomac, 2019-06-12:
@fsphil  : in France there are people who collect old secam TV sets (built before 1980), these TVs work only with field identification, 

on these TV the only solutions for displaying color are to use old VHS secam cassettes (from 1970's/80's) or hardware which converts scart signal (secam video composite) to secam L with color field identification, not easy to find on ebay and the price can be high :
http://dicovisuel.blogspot.com/2015/10/adaptateur-peritel-cgv-canal.html

I think a software solution with hackTV/hackRF would be great, and maybe not difficult to implement in the source code (just add "9 bottles" signal on one  VBI line of the secam signal)

closed 2019-09-21 by fsphil

Potomac, 2019-09-24:
@fsphil : here is a python program with interesting algorithms about pal, secam, ntsc, d2mac and filters : 
https://github.com/kFYatek/color_modem

perhaps you can re-use some algorithms here to improve secam (and other standards) in hacktv