fsphil/hacktv #7: RAW Teletext Stream as Imput

veso266, 2018-06-15:
I was wondering if it would be possible to add RAW Teletext imput into stdpipe like rasp-teletext is doing: https://github.com/ali1234/raspi-teletext
https://github.com/ali1234/raspi-teletext/blob/master/teletext.c#L154
https://github.com/ali1234/raspi-teletext/blob/master/buffer.c#L78 //read_packets() is here

so I could use https://github.com/peterkvt80/vbit2 to generate Teletext Data from tti files and then teletext.c to just modulate them

maybe you could also borrow some code from raspi-teletext to add Closed-Captions support for NTSC: https://en.wikipedia.org/wiki/EIA-608
https://github.com/ali1234/raspi-teletext/blob/master/cea608.c
also don't forget the RAW imput so I can use https://github.com/peterkvt80/newfor-subtitles to generate CC and your teletext.c to modulate them

peterkvt80, 2018-07-06:
I don't know about the hacktv parser but what vbit does is it makes a list of pages and sorts them into magazines. It then plays out those magazines in parallel mode. It applies a weighting to each magazine to adjust the bandwidth share allocated. I think that there is also packet 8/30 being transmitted at the appropriate intervals. So it is pretty close to UK broadcasting practise.

fsphil, 2018-07-08:
I've just added support for the raw 42 byte packets. Raw sources are indicated with the raw: prefix. Example:

`vbit2 --dir teefax | hacktv -f 855250000 -g 47 --teletext raw:/dev/stdin test`

commit 89124ce6 (2018-07-08): Add raw teletext packet support. Fixes #7.
closed 2018-07-08 by fsphil