fsphil/hacktv #9: Add Support To Decode The Raw

chezsick, 2018-09-11:
Hello My Name Is Sick, I Never Found To Read The Raw Output From hacktv, And I See Zcooger To Decode The Raw From HackTV In The Comment [On Youtube](https://www.youtube.com/watch?v=Av0TJ7Ajtcc&lc=Ugyp1f6wtdWljXfboTh4AaABAg.8iSeovRNYsI8iSlQ8I0daC) But Only On Windows

So, Is Possible To Add Option For Example -d , To Decode The Raw Input (Output From HackTV) And The Output Is A Format Readable To Any Players?

Thanks, Sick

fsphil, 2018-09-11:
The output from hacktv is not meant to be viewed directly, but with the correct configuration it can be done. For example:

`# hacktv input.mp4 -m pal -t uint8 -s 13500000 -o /dev/stdout | mplayer -demuxer rawvideo -rawvideo fps=25:w=864:h=625:y8 -`

closed 2018-09-11 by fsphil

fsphil, 2018-09-11:
hacktv is just an encoder, it can't decode a signal. But you already have the original file, just play that?

chezsick, 2018-09-11:
Yes, But With Mplayer, How To Render In Color

Zcooger, 2018-09-26:
PAL color decoder can be based on this program: http://www.techmind.org/vd/paldec.html
Open source code: https://drive.google.com/open?id=1eEjtenbwYG0KiICqwRa285S5HqdccKIn
But you will have to find someone who will modify it to work with video files for you.

veso266, 2018-12-21:
@Zcooger still have the source for PAL color decoder?

Zcooger, 2018-12-21:
Hello, it's there:
[palsrc.zip](https://github.com/fsphil/hacktv/files/2704380/palsrc.zip)