fsphil/hacktv #11: How to use?

MahmoudThePeltist, 2018-09-25:
After testing the program using this example: **$ hacktv -f 551250000 -m i -g 47 test**, i tried to transmit an image like this **$ hacktv -f 551250000 -m i -g 47 image.jpg**, but it failed, telling me that i was using a depreciated pixel format. and told me that it could not find an audio stream.

Then i tried to test with a video, **$ hacktv -f 551250000 -m i -g 47 video.webm** but that failed too, the issue seemed to be the video codec.

I also converted the video into a signal using the example **$ hacktv -o baseband.bin -m pal video.webm** and then tried to transmit it like this **$ hacktv -f 551250000 -m i -g 47 baseband.bin** but that also failed.

If you could please add an example on how to transmit a video/image and what the requirements are for said video/image, i would be very grateful, my final goal is to be able to use this to stream a video from my webcam, but first a gotta be able to send at least a picture lol

fsphil, 2018-09-26:
Hi! There are several things going on here...

- hacktv can't (yet) transmit a single image file. This is something I'd like to add, but unsure how best to do it.
- It also can't read its own output. hacktv uses ffmpeg to read and decode video files, but baseband.bin will contain raw RF data that means nothing to ffmpeg.
- Your example `$ hacktv -f 551250000 -m i -g 47 video.webm` is correct, but depending on the version of ffmpeg you have there may be some issues with certain file formats. Are you able to play an MP4 file?

fsphil, 2018-10-30:
There's no way to read directly from a webcam yet, but it is something I'd like to add eventually. It might be possible to do it now by using ffmpeg to create a stream that hacktv can play.

closed 2018-10-30 by fsphil