fsphil/hacktv #12: Use on old TV standards

Murphyv310, 2018-09-27:
Hi.
I know some are trying to use this as a System A modulator for 405 lines and sound. 
The 405 line system uses positive modulation on AM and audio on AM at 3.5MHz below the vision carrier. In other words two Hack RF devices would be required as System A doesn't have inter carrier sound. Also standards conversion for 405 lines would be required,  the Hack RF is not a standards converter so this has to be programmed in the PC.

fsphil, 2018-09-27:
System A should already work on hacktv with audio, however I don't have a 405-line TV to test it and I haven't heard of anyone else trying.

Only one hackrf is needed. Both the audio and video together fit well within the hackrf's 20mhz bandwidth.

Murphyv310, 2018-09-27:
Thanks Philip.
I've now had a better look through the documentation and I see where I have tripped up, apologies for that. I'm still not sure if I should buy a HackRF for 405 line and 240 line generation, adding in lines of code is not one of my strong points. I use Linux Mint and also Windows 7, any advice welcome.

fsphil, 2018-09-28:
I'm not sure. The hackrf is fairly expensive and can be a bit fragile (I've already broken the output amplifier in mine), but it's very flexible. What alternatives are there for transmitting 405 and 240 line TV?

closed 2018-10-06 by fsphil