fsphil/hacktv #14: how to use NICAM stereo

veso266, 2018-12-21:
Hi was wondering since you added NICAM stereo :), how would I use that?, I found nothing in hacktv.c only experimental VSB modulator option  (is that nicam)

also was wondering if you will support NICAM dual audio? (I was in indonesia, although they are switching ananlog off Bali still has it and instead of teletext they have Nicam there also in multi-language mode)

also here are the instructions I wrote for myself (because I forget stuff) on how to compile hacktv with Mysys2 (MinGW) on windows

1. Install FFMpeg-Dev
	1. Download FFMpeg Dev-Libs: https://ffmpeg.zeranoe.com/builds/win64/dev/ffmpeg-20180608-2bd26de-win64-dev.zip
	2. Copy *.pc files found in hacktv/pkg-config-files to msys64/mingw64/lib/pkgconfig
	3. Copy *.dll.a files found in ffmpeg-20180608-2bd26de-win64-dev/lib to msys64/mingw64/lib 
	4. Copy *.* found in ffmpeg-20180608-2bd26de-win64-dev/include to msys64/mingw64/include

2. Install LibHackRF
	1. Find Precompiled libHackRF somewhere (or compiie it yourselif: http://hellmund.blogspot.com/2015/05/hackrf-on-windows-8.html)
	2. Copy *.pc files found in hacktv/pkg-config-files to msys64/mingw64/lib/pkgconfig

2. Compiling LibHackRF
	1. Download Precompiled FFTW: http://fftw.org/install/windows.html
	2. Download libusb1.0: https://github.com/libusb/libusb/releases/download/v1.0.22/libusb-1.0.22.7z
	MinGW64/dll (where we need our dll)

	2. git clone https://github.com/mossmann/hackrf.git
	3. Compiling 
	mkdir host/build
	cd host/build
	cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/home/Mitja/libs/libusb/include/libusb-1.0 -DLIBUSB_LIBRARIES=/home/Mitja/libs/libusb/MinGW64/dll/libusb-1.0.dll -DFFTW_LIBRARIES=/home/Mitja/libs/libfftw/fftw
NO	cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=../../../../libusb/include/libusb-1.0 -DLIBUSB_LIBRARIES=../../../../libusb/MinGW64/dll/libusb-1.0.dll -DFFTW_LIBRARIES=../../../../libfftw/fftw
	make
	make install

4. Compiling HackTV
	Modify hackrf.c
		rename _close and _write to _closeRF and _writeRF
	Modify teletext.c
		rename gmtime_r to gmtime_s

NF:-1,CMC California (720p)
http://cmctv.ios.internapcdn.net:80/cmctv_vitalstream_com/live_1/CMC-TV/CCURstream.m3u8
#EXTINF:-1,CMC Country California (480p)
http://cmctv.ios.internapcdn.net:80/cmctv_vitalstream_com/live_1/CMCUSA/CCURstream.m3u8

Example Command to play CMC Country Television on Channel 31 with teletext
hacktv -f 551250000 -m b -G 1 -a -g 47 --teletext teefax http://cmctv.ios.internapcdn.net:80/cmctv_vitalstream_com/live_1/CMCUSA/CCURstream.m3u8

Test card and Teletext from vbit2: vbit2 --dir teefax | hacktv -f 855250000 -g 47 --teletext raw:/dev/stdin test

hope someone will find them usable

fsphil, 2018-12-22:
NICAM stereo is enabled by default for PAL I, B and G.

Dual-mono is tricky as hacktv is build around there only being one single stereo channel. I'd love to try it sometime but it's not a priority at the moment.

closed 2018-12-22 by fsphil

veso266, 2018-12-22:
also I tried to use it with vbit2 on windows like this: vbit2 --dir teefax | hacktv -f 855250000 -g 47 --teletext raw:/dev/stdin test

but windows does not have /dev/stdin so I modified teletext.c from this s->raw = fopen(path + 4, "rb"); into s->raw = fopen(stdin + 4, "rb"); but for some reason hacktv still complains: Unable to initialise video encoder.

and I got some warnings when compiling 
``c
teletext.c: In function 'tt_init':
teletext.c:1081:18: warning: passing argument 1 of 'fopen' from incompatible pointer type [-Wincompatible-pointer-types]
   s->raw = fopen(stdin + 4, "rb");
                  ^~~~~
In file included from teletext.c:33:0:
C:/msys64/mingw64/x86_64-w64-mingw32/include/stdio.h:600:17: note: expected 'const char * restrict' but argument is of type 'FILE * {aka struct _iobuf *}'
   FILE *__cdecl fopen(const char * __restrict__ _Filename,const char * __restrict__ _Mode) __MINGW_ATTRIB_DEPRECATED_SEC_WARN;
``
not sure what I am doing wrong here

fsphil, 2018-12-28:
I opened a new issue for that, please check that for a possible solution.