fsphil/hacktv #16: Support For TV Software In Local (ex. tvtime)

chezsick, 2019-01-02:
hello, i can't read the raw form hacktv, to a software in local for example tvtime reading the /dev/video0
is possible? thank Sick!

fsphil, 2019-01-02:
Are you trying to visualise the raw data? If so, try something like this:

`$ hacktv -m pal -t uint8 -s 13500000 -o - test | mplayer -demuxer rawvideo -rawvideo fps=25:w=864:h=625:y8 -`

chezsick, 2019-01-02:
Ok And The Size Other Mode, What Size Frame To Correct Read, NTSC ...

fsphil, 2019-01-02:
Look for the frame rate and full frame resolution in the hacktv output:

```
$ ./hacktv -m ntsc -t uint8 -s 13500000 -o /dev/null test
Video: 715x480 29.97 fps (full frame 858x525)
```

To view the raw output for the above in mplayer, you need to use 858x525 and 29.97 fps:

`$ hacktv -m ntsc -t uint8 -s 13500000 -o - test | mplayer -demuxer rawvideo -rawvideo fps=29.97:w=858:h=525:y8 -`

chezsick, 2019-01-02:
End And The Old TV No Color, What Size Frame?

chezsick, 2019-01-02:
The Other Mode, What Is The Size Frame?

fsphil, 2019-01-02:
Same method as above.

closed 2019-01-02 by fsphil