fsphil/hacktv #17: Bug in Syster-Encoder?

neo7530, 2019-01-13:
Hi,
is there any bug in the syster encoder? In the decoded picture i have many stripes / Lines jumping arround on wrong positions... Have tested per live-view in MeuhMeuh-TV aswell the decoded baseband-file per offline descrambler (cryptimage). There all the same stripes in the top of the picture. I have tested my own encrypded videos, they didn't have any stripes in view.

Regards, Marco.

Zcooger, 2019-01-13:
There is no bug. Some capture cards send the picture with subtle line shift, it means 2 lines may be added on the bottom so you have to adjust the shift in MMTV settings.

neo7530, 2019-01-13:
but when i decode the baseband manually, then no capture-card is involved, so it have to be clean i think....

Zcooger, 2019-01-13:
Is your baseband file made in 626 lines or 625?

neo7530, 2019-01-13:
625

neo7530, 2019-01-13:
https://gtrx1.selfip.net/stripes.jpg look @ this

Zcooger, 2019-01-13:
Send them both maybe

neo7530, 2019-01-13:
it tooks a moment, have to boot my linux-pc ...

neo7530, 2019-01-13:
https://gtrx1.selfip.net/clear.png it is a clear image

Zcooger, 2019-01-13:
While trying to decrypt in MMTV you can try to adjust bad picture shift caused by capture card itself or it's driver before it goes to the decoder - try all possible values closest to 0:
[image]
Cryptimage baseband decoding is not perfect too.

neo7530, 2019-01-13:
https://gtrx1.selfip.net/sys0000.png baseband decoded syster-frame

neo7530, 2019-01-13:
Okay, will give it a try tomorrow... Thank you anyway.

neo7530, 2019-01-13:
aahhhrrrgghhhhssss... -2 was the magic number.... now all the stripes are gone...

closed 2019-01-13 by neo7530