fsphil/hacktv #22: Autoplay feature?

SAUCYSOUP76, 2019-01-27:
Could an autoplay feature be added? Would like it to play all the videos in my directory.
Thanks

fsphil, 2019-01-28:
I've been considering adding a playlist function. It wouldn't be hard to generate a playlist for all the files in a directory.

For the EPG, do you mean having a UI on the video output?

SAUCYSOUP76, 2019-01-28:
For the EPG i meant a GUI themed application that you could choose when a video plays at a certain time of the day.

SAUCYSOUP76, 2019-01-30:
@fsphil got to say though this is a great piece of software been having fun with it!
Will you be adding a playlist function? If so when do you predict it'' be pushed out by?
I ask as it's a pain dragging and dropping each video file when the stream ends as well as the CRT making a loud pop at the end.

neo7530, 2019-01-31:
sorry, forgot the link: https://stackoverflow.com/questions/11647868/using-ffmpeg-to-stream-continuously-videos-files-to-a-rtmp-server

SAUCYSOUP76, 2019-01-31:
@neo7530 I will be honest i have no idea how to implement that into hacktv.

neo7530, 2019-01-31:
You have to write a batch / Shell Script and pipe the output into hacktv via stdin. Give me some time, i will test it tomorrow and will write a manual.

SAUCYSOUP76, 2019-01-31:
Also another possible feature, having compatibility with open broadcast studio.

neo7530, 2019-02-01:
for the playlist-feature: generate two txt-files:

files1.txt:
ffconcat version 1.0
file trailers.mkv
file a_movie.mkv
file trailer.mkv
file 'files2.txt'

files2.txt:
ffconcat version 1.0
file anothermovie.mkv
file commercials.mkv
file athirdmovie.mkv
file files1.txt

then generate a shell-command file:

eg: play.sh

ffmpeg \
-f concat -i files1.txt \
-c copy \
-f mpegts - | hacktv -f 799250000 -m i -g 47 -

execute the shell... while playing the content of 'files1.txt' you can edit 'files2.txt'...

SAUCYSOUP76, 2019-02-01:
@neo7530 getting command not found.

SAUCYSOUP76, 2019-02-01:
@neo7530 ran the command ffmpeg -f concat -i files1.txt -c copy -f mpegts - | hacktv -f 471250000 -m i -g 47 -
 as a command and says unsafe filename and files.txt: Operation not permitted

SAUCYSOUP76, 2019-02-01:
@neo7530 added -safe 0 before the -i and i now get this av_interleaved_write_frame(): Broken pipe
Error writing trailer of pipe:: Broken pipe

neo7530, 2019-02-01:
Have you wrote the full path in files.txt? /path/to/Video.mkv? Try to write files in Single quotes. '/path/to/Video.mkv'

SAUCYSOUP76, 2019-02-01:
@neo7530 yeah still getting broken pipe

SAUCYSOUP76, 2019-02-01:
@neo7530 file 'files2.txt' is it suppose to be in ' '? as i notice file files1.txt is different

neo7530, 2019-02-01:
Hmm, try it... I have to work.

wsygog, 2019-02-01:
I think it is better to leave playlist to ffmpeg pipe. Although for the user it might be a little complicated, example can be given easily, and ffmpeg has already done the hard work.

fsphil, 2019-02-03:
It's probably easier to use xargs: `$ cat playlist.txt | xargs -d "\n" ./hacktv -f 855250000 -g 47`
I'd still like a proper playlist function within hacktv, but this does the job for now.

captainjack64, 2019-02-05:
Alternatively, you can supply wildcards or limited regexp in filenames.

./hacktv -f 855250000 -g 47 somedir/[L-P]*.mkv

SAUCYSOUP76, 2019-02-06:
> It's probably easier to use xargs: `$ cat playlist.txt | xargs -d "\n" ./hacktv -f 855250000 -g 47`
> I'd still like a proper playlist function within hacktv, but this does the job for now.

Thanks that worked! Though would like it built into hacktv, but this will work for the time being.