fsphil/hacktv #25: Scrambled baseband output is delayed due to empty buffer lines

Zcooger, 2019-02-11:
This issue does not affect complex output at all - only file input where not every viewing application let the user to skip desired amount of data (header skip).
FTA:
[image]
Datacom Videocrypt - added one buffer line:
[image]
Nagravision Syster - added 344 buffer lines:
[image]

closed 2019-02-11 by fsphil

fsphil, 2019-02-11:
This is not a bug, hacktv's output is not intended for display like this. But it is handy to have the output aligned for testing, so I've applied a fix for it.

Your decoder software should be using the sync pulses to find the start of a frame rather than making assumptions.

Zcooger, 2019-02-16:
Hello, I applied your fixes to HackTV but the first baseband frame is still the same.

fsphil, 2019-02-16:
Looks fine here?
[image]

Zcooger, 2019-02-16:
Yup.

commit 2cc4ed12 (2019-02-11): Drop initial delay lines added by the video scramblers. Fixes #25