fsphil/hacktv #26: Problems when building HackTV for Windows (implicit declarations)

Zcooger, 2019-02-12:
System Info:
```
System:    Host: MINT-PORTABLE Kernel: 4.15.0-36-generic x86_64 bits: 64 gcc: 7.3.0
           Desktop: Cinnamon 3.8.9 (Gtk 3.22.30-1ubuntu1) dm: lightdm Distro: Linux Mint 19 Tara
Machine:   Device: laptop System: LENOVO product: 24295XG v: ThinkPad T530 serial: N/A
           Mobo: LENOVO model: 24295XG serial: N/A
           UEFI [Legacy]: LENOVO v: G4ETB3WW (2.73 ) date: 06/11/2018
           Chassis: type: 10 serial: N/A
Battery    BAT0: charge: 44.5 Wh 99.9% condition: 44.5/85.9 Wh (52%) volts: 12.2/10.8
           model: SANYO 45N1173 serial: <filter>status: Full
CPU:       Dual core Intel Core i7-3520M (-MT-MCP-) arch: Ivy Bridge rev.9 cache: 4096 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 11574
           clock speeds: min/max: 1200/3600 MHz 1: 1197 MHz 2: 1197 MHz 3: 1197 MHz 4: 1198 MHz
Graphics:  Card-1: Intel 3rd Gen Core processor Graphics Controller bus-ID: 00:02.0 chip-ID: 8086:0166
           Card-2: NVIDIA GF108M [NVS 5400M] bus-ID: 01:00.0 chip-ID: 10de:0def
           Display Server: x11 (X.Org 1.19.6 )
           drivers: modesetting,nvidia (unloaded: fbdev,vesa,nouveau)
           Resolution: 1600x900@60.22hz
           OpenGL: renderer: N/A version: N/A Direct Render: N/A
Audio:     Card Intel 7 Series/C216 Family High Def. Audio Controller
           driver: snd_hda_intel bus-ID: 00:1b.0 chip-ID: 8086:1e20
           Sound: Advanced Linux Sound Architecture v: k4.15.0-36-generic
Network:   Card-1: Intel 82579LM Gigabit Network Connection
           driver: e1000e v: 3.2.6-k port: 7080 bus-ID: 00:19.0 chip-ID: 8086:1502
           IF: enp0s25 state: up speed: 100 Mbps duplex: full mac: <filter>
           Card-2: Intel Centrino Advanced-N 6205 [Taylor Peak]
           driver: iwlwifi bus-ID: 03:00.0 chip-ID: 8086:0085
           IF: wlp3s0 state: down mac: <filter>
Drives:    HDD Total Size: 870.2GB (32.5% used)
           ID-1: /dev/mmcblk0 model: N/A size: 1.0GB serial: <filter>
           ID-2: /dev/sda model: Corsair_Force_LE size: 120.0GB serial: <filter>
           ID-3: USB /dev/sdb model: MK7559GSM size: 750.2GB serial: <filter>
Partition: ID-1: / size: 19G used: 9.4G (54%) fs: ext4 dev: /dev/sdb5
           ID-2: /home size: 208G used: 11G (6%) fs: ext4 dev: /dev/sdb7
           ID-3: swap-1 size: 2.05GB used: 0.00GB (0%) fs: swap dev: /dev/sdb6
RAID:      System: supported: N/A
           No RAID devices: /proc/mdstat, md_mod kernel module present
           Unused Devices: none
Sensors:   System Temperatures: cpu: 54.0C mobo: N/A gpu: 0.0:53C
           Fan Speeds (in rpm): cpu: 2388
Repos:     Active apt sources in file: /etc/apt/sources.list.d/official-package-repositories.list
           deb http: //mirrors.evowise.com/linuxmint/packages tara main upstream import backport
           deb http: //ubuntu.mirror.true.nl/ubuntu bionic main restricted universe multiverse
           deb http: //ubuntu.mirror.true.nl/ubuntu bionic-updates main restricted universe multiverse
           deb http: //ubuntu.mirror.true.nl/ubuntu bionic-backports main restricted universe multiverse
           deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           deb http: //archive.canonical.com/ubuntu/ bionic partner
           Active apt sources in file: /etc/apt/sources.list.d/spotify.list
           deb http: //repository.spotify.com stable non-free
Info:      Processes: 239 Uptime: 54 min Memory: 1739.6/3645.5MB
           Init: systemd v: 237 runlevel: 5 Gcc sys: 7.3.0 Client: Unknown python3.6 client inxi: 2.3.56

```
Log:
```
zcooger@MINT-PORTABLE:~/Desktop/hacktv-master$ sudo ./build_win64.sh 
[sudo] password for zcooger:         
+ HOST=x86_64-w64-mingw32
++ pwd
+ PREFIX=/home/zcooger/Desktop/hacktv-master/build_win64/install_root
+ export PKG_CONFIG_PATH=/home/zcooger/Desktop/hacktv-master/build_win64/install_root/lib/pkgconfig
+ PKG_CONFIG_PATH=/home/zcooger/Desktop/hacktv-master/build_win64/install_root/lib/pkgconfig
+ mkdir -p /home/zcooger/Desktop/hacktv-master/build_win64/install_root
+ cd build_win64
+ [[ ! -f /home/zcooger/Desktop/hacktv-master/build_win64/install_root/lib/libusb-1.0.a ]]
+ [[ ! -f libusb-1.0.22.tar.bz2 ]]
+ wget https://github.com/libusb/libusb/releases/download/v1.0.22/libusb-1.0.22.tar.bz2
--2019-02-12 15:33:24--  https://github.com/libusb/libusb/releases/download/v1.0.22/libusb-1.0.22.tar.bz2
Resolving github.com (github.com)... 140.82.118.4, 140.82.118.3
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
[... 1105 lines cut ...]
In file included from libusbi.h:565:0,
                 from core.c:44:
os/poll_windows.h: At top level:
os/poll_windows.h:61:2: error: unknown type name ‘OVERLAPPED’
  OVERLAPPED *overlapped;  // what will report our I/O status
  ^~~~~~~~~~
cc1: some warnings being treated as errors
cc1: some warnings being treated as errors
Makefile:657: recipe for target 'libusb_1_0_la-hotplug.lo' failed
make[1]: *** [libusb_1_0_la-hotplug.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
Makefile:650: recipe for target 'libusb_1_0_la-descriptor.lo' failed
make[1]: *** [libusb_1_0_la-descriptor.lo] Error 1
cc1: some warnings being treated as errors
Makefile:664: recipe for target 'libusb_1_0_la-io.lo' failed
make[1]: *** [libusb_1_0_la-io.lo] Error 1
cc1: some warnings being treated as errors
Makefile:643: recipe for target 'libusb_1_0_la-core.lo' failed
make[1]: *** [libusb_1_0_la-core.lo] Error 1
make[1]: Leaving directory '/home/zcooger/Desktop/hacktv-master/build_win64/libusb-1.0.22/libusb'
Makefile:476: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1
zcooger@MINT-PORTABLE:~/Desktop/hacktv-master$ 

```

neo7530, 2019-02-12:
I have setup a virtual Fedora distri for compiling it for Windows.

fsphil, 2019-02-16:
I can't really help with this, cross compiling is a bit of a dark art to me. It's not clear from the errors what's missing in your system.

Zcooger, 2019-02-16:
Ok. I will try next time in the future. I close it for now.

closed 2019-02-16 by Zcooger