fsphil/hacktv #27: How to mark a videoline with additional data?

neo7530, 2019-02-14:
Can you explain, how i can write the r and s values from the syster Encoder to the last line of each field as binary value. (the same way as cryptimages does) e.g: 01 for table1 00000110 for Offset and 10010101 for step as binary black&white pattern? can i use "vbidata_render_nrz()" and if yes, how to implement it?

Thank you.

Zcooger, 2019-02-15:
I think implementing CryptImage tag now is needless now. It needs complete field tagging revision to avoid any misunderstandings with WSS signaling which occupies the same area as the CI tag. I will talk with Mannix about it. Ideal solution would be placing single-line tag after the end of line 623 (bottom field, second half of the last line) - opposite side for WSS. Yes, reusing wss.c is straightforward and probably simple solution.

neo7530, 2019-02-15:
the wss line (1) never reaches line 288 or line 576, it will always be anywhere before line 576... for sure, it will be best to know how the real syster Decoder works, to put the data in the vbi area and patch cryptimage to work with real vbi data… 👍

fsphil, 2019-02-16:
Yes just copy the WSS code, it's quite close to what you want. You can modify it to write on any line you want.

Also if you have any information on the VBI data structure for Syster, please let me know :-)

closed 2019-02-16 by fsphil