fsphil/hacktv #29: PAL-FM changes

neo7530, 2019-03-28:
Have tried some things with PAL-FM (Analog sat receiver) and got nothing good out of hacktv. Thanks to captain_jack i made some changes in video.c

.fm_deviation   = 6000000, /* kHz */

and 

.video_level    = 0.51, /* Power level of video */

so i will have a video. But it is too bright. Gamma seems to have no effect to FM Video.

Zcooger, 2019-04-18:
In https://github.com/fsphil/hacktv/blob/0a08c326ffcde86af0a0d57a5a411d58ff003ab5/video.c#L196 gamma setting should be 1.0, so the image black and white balance is normal (linear gradient remains linear on oscilloscope).

fsphil, 2019-04-18:
Without a CCIR-405 pre-emphasis filter the FM video from hacktv will never look good, unless your satellite receiver has an option to bypass its de-emphasis filter.

closed 2019-09-21 by fsphil