fsphil/hacktv #32: Videocrypt conditional acsess mode

veso266, 2019-05-16:
Hi I saw you addded Conditional acsess mode for Video Crypt

I am wondering do you need MTV Sample Data or can you generate your own Data

and this Sky Cards do they self destruct after a while or if you have one that was disabled by your SAT provider would it still work

What is inside this card, can I use a normal recent SAT Card or does it have to be Sky

Thanks for Anwsering and forgive my stupidity and Best Regards

neo7530, 2019-05-16:
In this trunk, only a mtv smart card is supported. We don't have any smartcard for mtv, so we took 2 codewords and let a goldwafer card give the right response to the Decoder. The Cut points are used from a "codelist" to scramble the picture the right way. Captain Jack and me wrote some goldwafer and funcard Firmware to emulate an issue 07 smart card used by the adult channel in the late 90th... Captain Jack has implemented the Realtime cutpoint prbs in his trunk, so we have a real working videocrypt System, that works with Real Decoder, Real smart card and goldcard or funcard. Plus i have implemented an own algorithm to calculate the needed codewords for the videocrypt decoder. The easiest way to Start with is using a goldcard or funcard in a videocrypt 1 Decoder. All files needed are found in captain Jacks github fork.

fsphil, 2019-05-16:
The conditional example sends ECM messages captured from an MTV sample. No real reason for it, other than that was what I happen to have on VHS :-) Only an active Sky card of the correct age will work. Most will not sadly.

As neo7530 mentioned you could also use a card programmed to answer to those specific ECM messages, or implement your own messages intended for a custom card.

For decoding you can use a stand-alone Videocrypt 1 decoder (model SVA1) or a satellite receiver with integrated videocrypt decoder. hacktv's FM output is quite poor quality so I'd recommend the SVA1 - you can plug it into most TVs SCART socket and it'll just work automatically.

fsphil, 2019-05-16:
> Captain Jack has implemented the Realtime cutpoint prbs in his trunk

Did he now. Ahem :-)

neo7530, 2019-05-17:
Ahh, you have it also... Didn't saw that before… :-)

captainjack64, 2019-05-17:
> Captain Jack has implemented the Realtime cutpoint prbs in his trunk.
I added randomly generated seeds and algo used by Sky 07 series cards. Cutpoint PRBS calculation routine is forked from this repo and no credit should be given to me for this!

veso266, 2019-05-17:
Thanks would somehting like this: https://www.ebay.co.uk/itm/VIDEOCRYPT-DECODER-MODEL-No-SVA-1-WITH-SKY-VIEWING-CARD-USED-INSTRUCTIONS/401746305434?_trkparms=aid%3D888007%26algo%3DDISC.MBE%26ao%3D1%26asc%3D20140328180637%26meid%3D7b92dcb48cdc42e9b59109334400b6bf%26pid%3D100009%26rk%3D1%26rkt%3D1%26sd%3D293059902301%26itm%3D401746305434&_trksid=p2047675.c100009.m1982 work?

it also has a card with it so :smile: 

BTW: what does Active card mean how can Satelite company deactivate it (could hacktv do the same?) (its just a piece of plastic with a chip)

BTW: you think you could post a video of this working?

fsphil, 2019-05-17:
Yep, that's the one. That card /probably/ won't work, but there's no harm in trying :-)
Any Videocrypt 1 decoder will work in hacktv's free mode, even without a card inserted.

The chip on the card is a essentially a computer, it received messages from the broadcaster via the decoder which can tell it to be active or inactive. hacktv can send custom messages to the card, but the Sky cards will ignore them as they won't have the correct cryptographic signature. A custom card wouldn't have this limitation.

Captain Jack has a good video here showing it working: https://www.youtube.com/watch?v=FgoNPRX53zY

I don't believe there are any analogue satellite channels anywhere in the world now.

veso266, 2019-05-18:
Thanks

now where could I get a custom card to play with?
are there still any pirates that sell this cards?

what about Video Crypt 2 I heard this recievers could be modified into VC2

what chip does the decoder expect?

PS: How much wide is the stream, could I use universal LNB and feed the output into AirSpy and try to decode DVB-S2 with this without needing to buy a special Box (because you always had to return this SAT boxes and SAT cards to the provider)

also how are messages to the Sky card encrypted? (the private key would probably not work in the new Sky SAT Cards so someone has to have it and could hopefully share it)

PS: What about movies that requre a PIN?: https://www.youtube.com/watch?time_continue=42&v=y0LiCSz7h4U

regarding analog: http://www.sathint.com/search?custom=analog
there aphears to be some channels although I don't know if this site is updated

now the stupidest question of them all and please forgive me for asking it and if you could explain it in plain english?
How do you align the dish to the satelite of interest 

lets suppose I am located here: https://www.google.com/maps/place/45%C2%B059'44.1%22N+14%C2%B021'28.3%22E/@45.995578,14.3556723,17z/data=!3m1!4b1!4m6!3m5!1s0x0:0x0!7e2!8m2!3d45.9955778!4d14.3578609

and I have clear sky above my head, no trees and hills

and I am looking at west and want to align the dish to: 131.0° West or some other satelite that maybe I could actually try for start

what do I have to do?

I never had a change to play with Analog Satelite unfortunately (nobody I knew had Satelite and we had cable and now have IPTV so :cry: ) I did play with Analog terestrial however, those were some glorious days: walking with a portable TV in one hand and LP Antena in the other :smile: 

when Digital came it ruiend it all for me, TV DXing was gone as I was never able to recieve anything other than local stations anymore (using the same equipment)

I have a dish now (the smallest one that exists) its a bit hevier than LP antena so you can't realy carry it that easly in one hand for long time but only the dish the equipment and SAT cards had to be returned (it was digtal anyway)

Sometimes I drive to Bosnia to experience that joy again because they still have Analog there :smile:

PS: why does this dish: https://multicomstore.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/1/_/1.8_2.4-sat-dish-web_1.jpg

have LNB mounted diferently than this one: https://images-na.ssl-images-amazon.com/images/I/41GK81AS4FL._SX425_.jpg 

is it the size of the dish that tells how LNB should be fixed to it?

PPS: Can I use a normal dish to transmitt from? LNB should work in both directions (if it can convert 10 Ghz to 900Mhz it can probably do the oposite: convert 900Mhz signal back to 10Ght) right? LNB its just a Down converter?

fsphil, 2019-05-18:
OK, briefly:

- The custom cards require programming. That's not something I have experience with yet, maybe the others here can help. But yes, you can still buy these cards.

- VC2 is not supported by hacktv yet, but free-access mode should be there soon thanks to @captainjack64. Decoder for this are much harder to find than VC1.

- I don't know much about DVB-S2. Most satellite transponders are over 20MHz wide, too wide for an airspy or even a hackrf.

- Sky never documented their card message structure, for obvious reasons :-) I don't think they where encrypted, just signed to prevent someone modifying or generating their own.

- The PIN number would have probably been typed into the phone when "buying" a film.

- I didn't realise there where still analogue channels. That's cool! The footprint is over South America. Unless you live in these circles, I don't think you'll ever be able to receive this satellite: https://www.satbeams.com/footprints?beam=5739

- I normally just move the dish around until I get a picture :-)

- Yes it's usually size related. Small dishes tend to be offset fed to minimise the shadow of the LNB and the arm: https://en.wikipedia.org/wiki/Offset_dish_antenna

- LNBs are usually just down converters. I'm sure there are versions that can work both ways but they will be less common. I never use a dish to transmit, my hackrf is connected to a small vertical antenna or directly to the satellite receiver (via a DC block and attenuator to protect the hackrf).

neo7530, 2019-05-19:
Sky never documented their card message structure, for obvious reasons :-) I don't think they where encrypted, just signed to prevent someone modifying or generating their own.

Thats right, they are using some "keyed prng" to generate the Code Word for de-/crypting and generate some sort of signature out of this. Nothing special encrypted. My Implementation Use xtea algorithm as prng.

veso266, 2019-05-19:
now there must be some retired sky engeneers out there that maybe have some clues on what kind of pseudo random generator this could be, probably decoder has its implementation on the chip too

captainjack64, 2019-05-19:
Sky engineers don't have this detail. This system was patented by Thomson and is still likely a closely guarded secret under NDA. If you want to learn more about algos used by Sky then I suggest you get The Black Book by John McCormac. He explains these things in great detail for Sky 07 series and Sky 09 series cards. This book was the basis for generating valid control words for VC in my fork. You can examine fsphil's code for understanding how cutpoints are generated from those control words. D/D2MAC white paper floating on Google also explains cutpoint calculations for Eurocrypt, which shares many similarities with VC.

fsphil, 2019-05-19:
The decoder only has the PRNG for calculating the cut-points, which we've already figured out. The system was designed so that most of the security would be in the card itself. This way only the cards needed to be replaced if the system was ever compromised. This did happen a few times.

I don't think it would be worth reverse engineering the Sky cards when we can make our own.

closed 2019-05-19 by fsphil

veso266, 2019-05-23:
does someone have the black book?
version 1,3,4
I found version 2 here: http://cd.textfiles.com/hackersencyc/PC/CRYPTO/EURO-SCR.HTM