fsphil/hacktv #35: EIA-608 closed captioning for NTSC

Zcooger, 2019-06-07:
Hello. This ticket is made to suggest adding closed captioning signal (EIA-608) used to broadcast subtitles for the transmission in NTSC system.

LD-Decode guys announce support for decoding the CC data: https://www.facebook.com/groups/2070493199906024/permalink/2345606389061369/
Example capture of NTSC subtitles in the picture: https://www.youtube.com/watch?v=I0y_BjqE26A
Technology Connections CC history video: https://www.youtube.com/watch?v=6SL6zs2bDks
Technical data: https://en.wikipedia.org/wiki/EIA-608

fsphil, 2019-06-07:
I'd like to add this at some point. I've only had a quick look, but so far I've been unable to find a copy of the specification.

ValZapod, 2021-05-14:
> but so far I've been unable to find a copy of the specification.

All CTA standards are free of charge now.

https://shop.cta.tech/products/line-21-data-services

I can download it for you.

fsphil, 2021-05-14:
I don't have a decoder yet, there is one on the way so hopefully I can have a play with this soon.

fsphil, 2023-11-07:
"soon" turned out to be two years. I've have had some success with EIA-608 on both NTSC and PAL. I'll try and get it published soon.... (but sooner than two years).

devfonks, 2025-04-15:
Any updates on this? I live in the US and have multiple CRT TVs and a couple new digital ones with analog tuners that can decode EIA-608, if testing is the only holdup. I've also been able to scrounge some info up on the waveform and bit stream characteristics, and I'd be more than willing to help contribute to the code from across the pond if the branch were published here.

fsphil, 2025-04-15:
As often happens I got distracted, but it's still within the two years I mentioned ;-)

Annoyingly the CC decoder I imported from the US seems to have failed completely, after being very intermittent.
If I where to try and get another one, is there a recommended decoder to get?

fsphil, 2025-04-15:
I've pushed the test code to my branch, use `--cc608` to enable.
It should display the caption "HELLO FROM LINE 21 & HACKTV!" every few seconds.

devfonks, 2025-04-15:
On Tuesday 15 April 2025 01:55:55 pm Philip Heron wrote:
> If I where to try and get
> another one, is there a recommended decoder to get?

As far as separate decoder boxes go, the NCI box that started it all is pretty 
much the gold standard, IIRC, the model number on that is the telecaption 
4000. Those hail from around 1980, if memory serves, but they made them for 
years.

As an alternative, every NTSC TV (sold in the US, but most likely available 
other places too) sold after 1997 has a closed captioning decoder built into 
the television which follows EIA-608. Some of these TVs won't have the 
teletext channel support but they are required at least to support the CC1 
and CC2 channels. If you happen to have an NTSC-M TV of this era, check and 
see if you can rummage in the menus to find the option to enable it. 
Multisystem flatscreens might also support it, but again, I'm not sure if 
that's as universal outside the US as it is here.

devfonks, 2025-04-15:
On Tuesday 15 April 2025 02:25:39 pm Philip Heron wrote:
> I've pushed the test code to my branch, use `--cc608` to enable.
> It should display the caption "HELLO FROM LINE 21 & HACKTV!" every few
> seconds.

I'll give this a test when I get back home tonight and report back if there's 
problems I can't fix :P

fsphil, 2025-04-15:
EIA-608 captions where used on some pre-recorded VHS cassettes in the UK but I don't think there was ever any TVs with integrated decoders. Broadcasters used Teletext to carry subtitles so I guess there wasn't much incentive.

In better news my decoder seems to have come back to life, mostly. The test captions are appearing.

fsphil, 2025-04-15:
My initial plan is simply to pass through EIA-608 data from any media file that has them (and if ffmpeg supports it):

[image]

devfonks, 2025-04-15:
That's what I was going to ask next, I got the captions to work on my tube,
and so the next step is streaming them from somewhere :P

I do know that EIA-608 can be streamed natively in MPEG containers, but I
don't know any details about it. AFAIK ffmpeg can be used to extract them
like any other stream in the muxed file. It seems that where it *possibly*
lacks support is actually embedding them, not that it matters for this
project.

If this gets off the ground I'm toying with the idea of creating a program
that can convert YouTube-style subtitle files into EIA-608 streams, which
could possibly be fed into hacktv in some way, either through embedding or
some other way (pipe? File?)

fsphil, 2025-04-15:
Indeed, the capture above is from a DVD with embedded EIA-608 subtitles (well, technically 708 subtitles?).
I have it mostly working but there's a few cases where it's corrupting the output.

fsphil, 2025-04-16:
This should be working now.

devfonks, 2025-04-16:
Awesome! I'll have to rip a DVD from my shelf to give it a test with streams. Did you just feed the VOB file directly into hacktv, or did you have to do some ffmpeg magic to get the streams right first?

fsphil, 2025-04-16:
I removed some of the extra audio tracks, but otherwise nothing else. The 608/708 captions are embedded into the video stream so as long as that hasn't been modified it should work.

devfonks, 2025-04-17:
[image]

I call that a success - having a look at the VBI on one of my more adjustable B&W sets visually shows the right signal levels and artificially degrading reception on this set seems to show the expected robustness. I've not run into any stream corruption yet with any of my DVDs.

fsphil, 2025-04-17:
Nice! Thanks for testing. I'll call that a victory. There's a bunch of other features that could be added but I'll lave that for another day.