fsphil/hacktv #36: AUDIO SYSTER does not encode audio

vanhoteen, 2019-06-19:
I'm happy with the software, but it encodes the video but the audio does not.
I have tried several mp2, mp4, acc, ac3, mp3 files
I have ubuntu and hackrf.
Can you help me?

vanhoteen@ubuntu:~$ hacktv -m g  -f 583000000 -g 47 /home/vanhoteen/Vídeos/canalplus1990.mp4 -r --syster
Video: 832x576 25.00 fps (full frame 1024x625)
Sample rate: 16000000
UOpening '/home/vanhoteen/Vídeos/canalplus1990.mp4'...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/vanhoteen/Vídeos/canalplus1990.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isommp42
    creation_time   : 2015-08-28T06:01:56.000000Z
  Duration: 00:02:39.57, start: 0.000000, bitrate: 318 kb/s
U    Stream #0:0(und): Video: h264 (Constrained Baseline) (avc1 / 0x31637661), yuv420p, 480x360 [SAR 1:1 DAR 4:3], 219 kb/s, 25 fps, 25 tbr, 25 tbn, 50 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 95 kb/s (default)
    Metadata:
      creation_time   : 2015-08-28T06:01:56.000000Z
      handler_name    : IsoMedia File Produced by Google, 5-11-2011
Using video stream 0.
Using audio stream 1.

fsphil, 2019-06-19:
Do you get audio without the --syster option?

vanhoteen, 2019-06-19:
if I have audio but it is not encrypted, it sounds perfect not like when it goes through the system nagra

fsphil, 2019-06-20:
Your decoder is probably expecting the audio to have been inverted. hacktv doesn't do this yet.

vanhoteen, 2019-06-26:
thanks ,I thought that if, seeing that video https://youtu.be/CML8Hr_eFZ8 , sorry my answer was late I was sick

Zcooger, 2019-09-13:
@shafau Waving sound volume is caused by sample rate differences between the video clock and the modulated audio. It has somehow to be synchronized, Nagravision terminal synchronizes the audio chip with video line rate.

fsphil, 2019-09-13:
Just after the call to _av_read_audio in video.c.

Don't exclude the possibility that the inversion frequency is not exactly 12.8 kHz, a slight offset could cause the beating mentioned.

shafau, 2019-09-15:
For now we have got something like that:
ffmpeg -i input.wav -f lavfi -i "sine=frequency=12800" -filter_complex "[1]volume=+18.064dB[sine];[0][sine]amultiply[mixed];[mixed]lowpass=f=12800" output.wav

We don't know what to do next and how insert that into hacktv

captainjack64, 2019-09-15:
If you have to use ffmpeg then you'd need an ffmpeg audio filter implemented in HackTV. It's probably better to do this post ffmpeg...

fsphil, 2019-09-19:
I've added support for audio inversion, use the --systeraudio flag to enable it.

closed 2019-09-19 by fsphil