fsphil/hacktv #45: WSS tag is erroneously overlayed over image scrambled with Videocrypt

Zcooger, 2019-09-13:
WSS tag is stationary.
Additionaly it is placed on wrong line since single line buffer is applied and in result it should land higher.
I bet aspect ratio switch is working until the video is scrambled. After decoding WSS tag may be displaced and messed by the decoder so I suggest to:
1. Overlay WSS tag first.
2. Scramble the image.

[image]

fsphil, 2019-09-13:
No, this is the correct behaviour. Videocrypt decoders do not modify line 23.

fsphil, 2019-09-13:
Normally videocrypt encoders will place the scrambled version of line 24 on line 23. hacktv doesn't do this in order to preserve the WSS signal. The result of this is that the WSS will actually appear twice in the descrambled output, an untouched version on line 23 and a 'descrambled' version on line 24.

closed 2019-09-13 by fsphil