fsphil/hacktv #46: --wss switch erases second half of picture information on line 23

Zcooger, 2019-09-13:
The whole line contains only the tag.
Example showing this error:
[image]
Example showing complete 23rd line:
[image]

fsphil, 2019-09-13:
The WSS signal overlapped the video on the second half of line 23 so I just blanked it completely.
It's difficult to say if this is an error, the specification ([EN 300 294](https://www.etsi.org/deliver/etsi_en/300200_300299/300294/01.03.02_60/en_300294v010302p.pdf)) doesn't mention video at all. There is just this:

> For optimum decoder performance, it is recommended that this period is free from other signals.

Zcooger, 2019-09-13:
Ok, I just base this ticket on my observations of my 3 DVD players and a DVB-T receiver output.

fsphil, 2019-09-14:
Do they all use the same timing? Can you measure how many microseconds it is from the start of the line?

Zcooger, 2019-09-14:
Sure. I will check it on every device I have.

fsphil, 2019-09-14:
Ah wait, I think I found it. It should be 42.5μs

closed 2019-09-14 by fsphil