fsphil/hacktv #47: VPS/PDC aka ShowView broadcast signaling

Zcooger, 2019-09-17:
This ticket is made to suggest adding VPS/PDC signaling known as the ShowView system.
It uses 16th line on PAL signal only.
Its passive function delivers TV station name to the receiver. Thanks to this signal some of TVs and recorders can tag auto-tuned channels using TV station's name.
The second passive function is to provide information about the start time of a television programme.
The active function broadcasts program ShowView code used by recorders to distinguish start, broadcast break and the end for easier recording. Any broadcast delays are compensated by VPS/PDC.

http://www.ti.com/lit/an/slea017/slea017.pdf

Zcooger, 2019-09-17:
btw... where's your PayPal?

fsphil, 2019-09-18:
The spec for this is here: https://www.etsi.org/deliver/etsi_en/300200_300299/300231/01.03.01_60/en_300231v010301p.pdf

The line 16 data doesn't include the channel name. It looks like that's taken from teletext, which hacktv already sends. It's fixed to "hacktv" at the moment; none of my TVs use it.

I'm not sure how PDC would work with hacktv? A program code could be set on the command line but it wouldn't change for the entire run. It would need to be integrated with some kind of scheduling system.