fsphil/hacktv #49: Payview III

justaconsumer, 2019-10-06:
Here's a idea for the HackTV-thing: Payview III.
I wish I had technical documents but I don't and I dunno where I can find any of this.

According to Wikipedia:
Used by German/Swiss channel Teleclub in the early 1990s, this system employed various methods such as video inversion, modification of synchronization signals, and a pseudo line delay effect.

I dunno if HackTV let alone HackRF can support modification of video signals. I dunno,

NOTE: I have only seen and heard about HackTV on videos around the Internet. I don't have a HackRF. I only got a RTL-SDR.

Zcooger, 2019-10-06:
Any documentation would be helpful. Any television signals can be created using HackTV but they have to be defined and more importantly physical decoder/attached cable box must exist and recognize these signals.

justaconsumer, 2019-10-06:
lemme guess... hacktv can actually manipulate signals as they're being transmitted?

Zcooger, 2019-10-06:
HackTV is ran by a single command, if you want to change any parameters you have to close it manually and start with new arguments. Of course it can generate ANY signals you define.

justaconsumer, 2019-10-06:
Nice.

For your last comment regarding documentation... I'm positive it's probably a safe-guarded secret [and probably still is]. This would be fine.. if I had video footage.

fsphil, 2019-10-06:
The official documentation would be nice, but yes that'll almost certainly still be a secret (or just filed away in storage and forgotten). In some cases the documentation written by people trying to hack the system is enough. A working decoder would be nice too as I'm reluctant to add scrambling modes that can't be used on real hardware.

justaconsumer, 2019-10-06:
I see.

Checking the biggest Swiss auction site for "Teleclub-Decoder" yielded 0 results on both ricardo.ch and eBay.ch.

I dunno why there are no boxes on auction but my only solution for this is to wait for any videos of Payview to surface, let alone any documentation. If there is a auction for such boxes or documentation someone found and told me about, I might bump this.

philpem, 2019-12-23:
According to McCormack, it's a random line inversion system with a very high video bandwidth compared to Astra. The encoding scheme apparently involves superimposing a high-amplitude signal on top of the 

Details of a pirate decoder can be found here: http://www.geocities.ws/ResearchTriangle/9405/tc.htm

It seems that the system that decoder decodes is described in patent GB2042849A: https://patents.google.com/patent/GB2042849A

There's also a later version of Payview which is described in EP0341801A2, which is more in line with the system described on Wikipedia: https://patents.google.com/patent/EP0341801A2/

Going by Google, it's also described in one of McCormac's books, "World Satellite TV and Scrambling Methods: The Technicians' Handbook" -- https://www.amazon.co.uk/World-Satellite-Scrambling-Methods-Technicians/dp/0917893190