fsphil/hacktv #51: Vertival Interval Time Code (VITC)

Zcooger, 2019-10-22:


fsphil, 2019-10-22:
Do you have or know of any devices that can make use of the feature?

Zcooger, 2019-10-23:
I don't have any Betacam SP or any professional gear to test that with but I found such signal on this repair demo of Sony PVW-2800 P so there is one example:

[image]

https://www.youtube.com/watch?v=HDQB1Ck_UjY

Besides this Cassette deck there were in use timecode inserters, extractors and counters. Some displays can use OSD to show it.
I found it also on my VHS tapes with live TV recordings where it slipped to the VBI during broadcast like here:

https://upload.wikimedia.org/wikipedia/de/3/34/Vitc-PAL.jpg
[image]

Now I found software that reads TC out of digitized video frames: https://www.videotoolshed.com/old/vitc-reader/

03stevensmi, 2023-01-12:
@fsphil @Zcooger @captainjack64 I would like to add that this is something I actually need to be able to make use of my old videomatte vm1 video mixer. I unfortunately can't use havktv because old video equipment like mixers etc need tbc vertical locked signals otherwise it just messes destroys the composite signal when in use with many of these old devices. Could you please add this feature?

fsphil, 2023-01-14:
I've pushed a VITC inserter, you can try it with `--vitc`. It should work with NTSC and PAL modes, but I have only been able ot test it with ffmpeg's `readvitc` filter. I'd appreciate some testing with real hardware if anyone has it.

fsphil, 2023-01-14:
> otherwise it messed up destroys the composite signal when in use with many of these old devices

This may still be an issue - right now hacktv doesn't set the colour framing bit. You might need this set, it's at https://github.com/fsphil/hacktv/blob/master/vitc.c#L136

fsphil, 2023-01-14:
One potential issue, the field flag seems to be in a different location for 25 fps modes vs. others. Right now it's fixed at the 25 fps position.

03stevensmi, 2023-01-15:
> One potential issue, the field flag seems to be in a different location for 25 fps modes vs. others. Right now it's fixed at the 25 fps position.

have you (or anyone here) got a compiled windows build i can test?
just set up eveerything and i am ready to give it a shot.

Thanks fsphil for all your work.
You have just saved my ass on this one let me tell ya ;)
once i get a hold of a windows build i will let you know if it works.
thank you!

03stevensmi

fsphil, 2023-01-15:
No worries. I'll close the issue for the moment. If you or anyone else are able to test it, drop a note here!

closed 2023-01-15 by fsphil