fsphil/hacktv #53: Show full options of hacktv when typing hacktv --help

Potomac, 2019-12-01:
Hello,

I notice that when we type "hacktv --help" then the output will not show the full features of hacktv,

for example if the user has not yet bought or connected a fl2k device then the "--help" command will not show the fl2k options, it's because Fsphill has restricted the information by adding a "#ifdef HAVE_FL2K" statement in hacktv.c file (line 123) :

https://github.com/fsphil/hacktv/blob/master/hacktv.c

I think it's a mistake, because usually a help file must by default show all options/features of the software, there is no need to restrict the information, a user must know all features of the software, especially if there is no PDF/html documentation,

I notice also that the readme file on github page doesn't mention the fl2k feature.

Here is in attached file what the user can read if he types the help command [hacktv-help.txt](https://github.com/fsphil/hacktv/files/3908657/hacktv-help.txt)

Potomac, 2019-12-01:
You can also create 2 type of help :

- a short help for displaying basic commands, a kind of help for beginners
- a long help, which display all commands, with examples (switches :  --full-help, --long-help)

I created this bugreport because I know someone who is a beginner about linux, he has installed/compiled with difficulties hacktv on a raspberry pi, and now he is trying to understand how to use hacktv with a fl2k device, 

it's not easy with the current help file (missing example about fl2k)

fsphil, 2020-08-05:
Documentation is definitely a weakness at the moment (as is timely response to github issues, sorry about that). The help text will now show all options regardless of the build configuration.

I'd like to write a bit about the various output devices, input sources, maybe the unique/special TV modes. I enabled the wiki option on github with the intention to start, but it never went anywhere.

ejelly, 2020-08-05:
Hang on, if it's really guarded by `#ifdef HAVE_FL2K`, then it has nothing to do with whether the user "bought or connected" an fl2k device, but rather whether that feature has been compiled into the binary?