fsphil/hacktv #56: SoapySDR?

neo7530, 2020-01-06:
Hi,
i have an adalm-pluto now and would like to use it with hacktv. How is the syntax for using it?

fsphil, 2020-01-06:
At a minimum:

`$ hacktv -o soapysdr`

Assumes you have the soapysdr driver installed correctly.

neo7530, 2020-01-06:
yep, that is okay. didn't know the device-args :( "tried -o soapysdr="device=plutosdr,ip=192.168.2.1" so hacktv will run, but no Output from Pluto...

fsphil, 2020-01-06:
You may need to select an antenna. I don't have one so I can't really help - I do believe it works though.

fsphil, 2020-01-13:
Any progress?

neo7530, 2020-01-16:
The pluto will be found if I start hacktv as root, but it won't transmit anything. I can see the frequency setup in spectrum monitor via hackrf, then no more carrier in rf spectrum.

fitzdoingprojects, 2020-01-18:
I'm having some success using hacktv with the plutosdr. I've managed to transmit a glitchy NTSC stream on UHF channel 28 with the following:

hacktv -o soapysdr -f 555250000 -s 1000000 -v -m m -g 89 test

similarly, I get a glitchy stream with a test.mkv video:

hacktv -o soapysdr -f 555250000 -s 1000000 -v -m m -g 89 test.mkv

Make sure that the plutosdr registers with SoapySDR, it should show up with the following command:
SoapySDRUtil  --probe

I've tried playing around with a few sample rates and it seems that the plutosdr only works with HackTV at 1Msps. The plutosdr should work all the way up to 10Msps. Do you have any thought on why this might be?

vdveer, 2020-02-15:
The soapysdr build checks did work for me running ubuntu 18.04 (with all the required soapy tools and dev libraries installed). My experiences with the soapysdr (running with HackRF and LimeSDR) is that it takes an extra hit on the CPU (uses 1 thread) compared to when I run it with a HackRF connected, so maybe it is cpu-related on your machine as well. I was using a Intel i5 8500 CPU at the time, and using soapy with both devices I had 100% load on one thread and a glitchy image. Hackrf directly had <100% load and a clear image.

Extra parameters (for me: switching between devices) worked for me with the following syntax `-o soapysdr:driver=lime`.

sidenote: my experiences are with the hacktv sources of 2-3 months ago.

fsphil, 2020-02-15:
It sounds like the soapysdr output needs to be in its own thread. I had naively thought it wouldn't need to be.

closed 2023-11-07 by fsphil