fsphil/hacktv #57: NTSC FFMPEG

Commander-Crash, 2020-02-19:
I live in the states and most if not all the TV's use NTSC. I get a error when using "-m ntsc" rf hackrf_open(): Unsupported mode output type for this device. so I use the "-m m" and it plays the video but its in black and white and no sound or static for audio. I tryed adjusting the sample rate and freq of the channle no go. the Test bars show up ok with command "hacktv --samplerate 9000000 -f 572000000 -m m -g 47 test" on ch 31 in the states. Why is it that any video i play is not clear or has issues? is their something im missing? Any help would be appreciated Thanks

fsphil, 2020-02-19:
The unsupported mode is expected, "-m ntsc" generates a baseband composite signal which wouldn't make sense on a hackrf. The "-m m" mode is what you need for RF.

Your other command should work, though I'd increase the sample rate a little. Try this:

`hacktv -s 13500000 -f 572000000 -m m -g 47 test`

It's also possible your machine isn't able to feed the hackrf quickly enough. What spec of machine are you using?

closed 2020-08-05 by fsphil

Commander-Crash, 2020-10-01:
Sry i forgot all about this and gave up after a while. This was on a rpi4 4gb