fsphil/hacktv #63: File Output Format

fitzdoingprojects, 2020-03-01:
I am a bit confused by the file output format. I would assume that the following:

hacktv -o test.dat -t int32 -s 100000 -v -m m test

Would produce a file 'test.dat' comprised of complex IQ (2x int32) samples, with 100k sample stored per second. However, 'test.dat' is much too large and has no dependence on the '-s' parameter.

My ultimate goal is to have hacktv transcode multiple videos into ntsc IQ files. Then have gnuradio broadcast the videos as multiple 'channels' from a single SDR. Please let me know if anyone has suggestions on how to do this?

fsphil, 2020-03-01:
Your assumption is correct, it'll create a complex int32 file. The test screen never ends, you can't use that to compare output size. Try using a short video.

Also 100000 is much too low a sample rate for system M, it's only 6 samples per line. You'd need at least 10MHz. (Note: There's currently a bug that may produce the wrong colours at certain sample rates when using NTSC - 13.5MHz works well. I'm hoping to get this fixed soon)

I've never tried transmitting simultaneous channels, but that should work. A hacktv would definitely have the capacity for two channels, maybe three with a bit of filtering.

fitzdoingprojects, 2020-03-01:
Oops, my mistake, the file sizes do scale linearly with the sample rate '-s' option. Also, I was confused about the data type gnuradio expects for its 'complex' data type. I thought 'complex' was two int32 rather than two floats. (I was working a bit late last night). 

NTSC - 13.5Mhz does indeed work well.  I am able to successfully transmit a file saved with the following command:

hacktv -o intro_float_13_5M.dat -t float -v -s 13500000 -m m intro.mp4

Where 'intro.mp4' is a short video. I can share my simple gnu radio example that works with the limesdr-mini, if anyone is interested.

And thanks for all the work on this project! It is a great entry point into the world of software defined radio.

fsphil, 2020-03-01:
Is the audio OK? None of my TVs support system M audio so I've never been able to test it.

You should also be able to transmit directly from hacktv with the limesdr-mini, if you have soapysdr and soapysdr's limesdr driver.

fitzdoingprojects, 2020-03-02:
Yep, audio works great! And yes, I've had great success using the limesdr-mini directly with hacktv. If a soapysdr shows up with:

SoapySDRUtil --probe

Then it works with hacktv using the following (may want to modify frequency, antenna, and gain):

hacktv -o soapysdr -f 555250000 -g 40 -v -m m test

I've also had some success with the Plutosdr, although the data transfer rate over usb 2.0 isn't high enough for ntsc.

The reason I'm trying to go through gnuradio is to get multiple video streams transmitted out of a single sdr. I'm working on an art project to transmit video to old analog TVs and I'm trying to keep costs and power consumption down.

I figured it would be easy enough to take a few IQ files generated by hacktv, frequency shift, and then transmit them out of a limesdr-mini. I also thought it would be less computationally intensive to use already generated IQ file, perhaps allowing the use of raspi --with a lot of memory. I'm working out the details on gnuradio now. It's my first time with the software and it's a bit of a learning curve. 

I took a quick look through the hacktv code base and couldn't see an easy way forward for multiple streams. If you have any ideas, I would like to help.

fsphil, 2023-11-07:
The options `-passthru` and `--offset` can be used to pipe the output of hacktv into another instance of hacktv to transmit multiple channels, bandwidth permitting. There's an example in the README.

These where added some time ago and I meant to update this issue but missed it.

closed 2023-11-07 by fsphil