fsphil/hacktv #64: fl2k device not producing any output

ljones0, 2020-04-11:
Hello,

First of all I'm finding that hacktv works fine with the hackrf :-) . But I also wanted to try hacktv with a fl2k based device (lspci lists it as a fresco logic fl2000  device, id 1d5c:2000). 

I can confirm my device works as I'm able to transmit FM via osmo-fl2k and fl2k_fm. But hacktv gives me an error with the same device.

I do have the relevent dev packages installed but when I go to use hacktv with the fl2k device I get this error;

$ ./hacktv -o fl2k:0 -m -i -f 551250000 test

Video: 832x576 25.00fps (full frame 1024x625)
Sample rate: 16000000
Using 6 zero-copy buffers
libusb: error [op_dev_mem_alloc] alloc dev mem failed errorno 12
Failed to allocate zerocopy buffer for transfer 4
libusb: error [submit_bulk_transfer] submiturb failed error -1 errorno=2
Failed to submit transfer 0
Please increase your allowed usbfs buffer size with the following command:
echo 0 > /sys/modue/usbcore/parameters/usbfs_memory_mb
Requested sample rate 16000000 not possible, using 15999998.000000, error is -2.000000

I'm getting the same error on two systems I've tried - an x86 ryzen 5 (debian, kernel 5.4.0-4-amd64) and also on a pi 4 (raspbian, kernel 4.19.97-v7l+). 

No signal appears to be produced x.x

ljones

fsphil, 2020-04-11:
hacktv only generates baseband signals, it relies on the SDR device's IQ modulator to produce the actual RF signal. The fl2k doesn't have a modulator, your command will result in a baseband AM signal on the red and green channels. The frequency argument is ignored in this case.

You'd need an equivalent of fl2k_fm but for IQ signals. The output frequency would likely be limited to < 100 MHz.

ljones0, 2020-04-11:
drat! Oh well, ne'er mind. Might be worth closing this but just one thing - remember that loud buzz or no sound on system A/405 lines? I tried something other than an old vintage sony TV (VHF, System A) - a hand scanner (Yaesu VR500) and tuned it to the same frequency and demodulation (AM) as system A sound.  I get the same issue; I do get sound but with a buzzing sound as well.

ljones

closed 2020-08-05 by fsphil