fsphil/hacktv #66: Windows building

ea3iav, 2020-06-22:
Hello! I have followed the instructions in order to execute the script. But I am having difficulties to find the specific packages shown here

[image]

I get an error 

`Line37 mingw64-cmake: command not found`

any help on this? I did find similar packages and installed them, but not those specific that were mentioned above. I guess it might be the reason of the error I get

fsphil, 2020-06-23:
Installing those packages brings in ming64-cmake on Fedora 32, I can't be sure about other distros. The exact package name that contains the file is `mingw64-filesystem-113-1.fc32.noarch`.

However there are other errors even with it installed. I'm not sure what's broken yet.

ea3iav, 2020-06-29:
Working fine! Used you zip with the hacktv ported executable amd ot worked!

Zcooger, 2020-07-07:
For anyone interested here's HackTV exe (07.07.2020): 
[hacktv.exev07.07.2020.7z.zip](https://github.com/fsphil/hacktv/files/4881973/hacktv.exev07.07.2020.7z.zip)

SheepyChris, 2022-10-10:
Sorry to necro this issue, however I am currently attempting to cross-compile to Windows and unfortunately I keep stumbling across this issue:

`+ rm -rf hackrf-2018.01.1/host/libhackrf/build`
`+ mkdir -p hackrf-2018.01.1/host/libhackrf/build`
`+ cd hackrf-2018.01.1/host/libhackrf/build`
`+ mingw64-cmake -DCMAKE_INSTALL_PREFIX=/home/chris/Documents/hacktv/build_win64/install_root -DCMAKE_INSTALL_LIBPREFIX=/home/chris/Documents/hacktv/build_win64/install_root/lib -DLIBUSB_INCLUDE_DIR=/home/chris/Documents/hacktv/build_win64/install_root/include/libusb-1.0 -DLIBUSB_LIBRARIES=/home/chris/Documents/hacktv/build_win64/install_root/lib/libusb-1.0.a`
`./build_win64.sh: line 37: mingw64-cmake: command not found`

I do currently have mingw-w64 installed on my distribution (MX Linux 21), attemping to only run `cmake` instead complains about a missing CMakeFiles.txt.

steeviebops, 2022-12-19:
mingw64-cmake is Fedora-specific. You can get it building on other distros by replacing all references of mingw64-cmake with the following three lines:

````
cmake .. \
	-DCMAKE_SYSTEM_NAME=Windows \
	-DCMAKE_C_COMPILER=$HOST-gcc \