fsphil/hacktv #67: I have been playing with the test page, would you like me to do a PR for my changes?

gregoryfenton, 2020-07-07:
Hi FSPhil
I love the app, and have been playing with the test page.

Where previously you had to bitmap any text you wanted it is now simple to add text at any position, any size and colour.

I have also patched into the frame update callback code to add a real-time clock, again positionable anywhere.

If you like it I will tidy the code, commit it to my github and generate a pull request.

[image]

Best de M0ODZ
Gregory Fenton

pkurucz, 2020-07-08:
I'd def like to see this merged, though I have no say in things :)

fsphil, 2020-07-09:
Hah, that looks great. Love that it's CP437.

I'm honestly not sure what to do with the test page. I'm happy to add more options, but I'm not sure how best to do it. What do people want it to do?

gregoryfenton, 2020-07-09:
For me the test page definitely seems a bit of a missed opportunity. A quick one liner and you can be transmitting a test page, but a pain if you actually want to do something with the page.
Considering you have 1/25 second (or 1/30 second NTSC etc) between frames there is plenty of time to do things before redraw happens. I was thinking of creating a static background, memcpy'ing that and every frame memcpy'ing it back then doing things such as text, maybe controlled by a text file:
[text 10,20,0xFFFFFF,0x000000,2,1]This text will be at x 10, y 20 in white on black[/text]
[text 10,100,0xFFFFFF,0x000000,2,1]{@clock DD/MM/yyyy HH:mm:ss.ff}[/text]
The code I have now could do that with minor changes.
If I just have static text and the clock is always at the same position I could skip the memcpy on frame update (as it is now, memcpy is a pipe dream) which would mean that CPU usage would be incredibly low - it just writes 128 bytes per letter which in real terms is nothing.

ejelly, 2020-07-09:
I personally would like it, and think it to be actually technically very useful, if there were actual PAL and NTSC test patterns available, e.g.: https://en.wikipedia.org/wiki/Philips_PM5544#/media/File:Philips_PM5544.svg
Given that this immediately shows an indication of issues with both the PAL/NTSC encoding/decoding and with the transmission.

I usually work around this by generating a single-frame movie with ffmpeg, but it's not trivial to get it right (there are very specific lines and phases in the test pictures that need some considerable care to reproduce through ffmpeg).

captainjack64, 2020-07-09:
I like it! Additional test cards would definitely be useful with custom text in text boxes. The static cards would be very easy to add in ppm or bmp format and then overlay the text.

ejelly, 2020-07-09:
Yeah. Though if you want to add them as static images (instead of generating them), I think they'd have to be in the YUV color space for their precise phase, and in fixed (common PAL/NTSC) resolutions because of their line alternating parts. Not sure BMP can deliver the color part, but with PPM it's probably possible, given its flexibility.

captainjack64, 2020-07-10:
Static images pose a problem when used with HackTV's dynamic sample rates. They would need to be resized, which is nowhere near as precise as generating them on the fly.

Another option is to use vectorised image formats (such as SVG above) - much smaller sizes and will fit any resolution. The disadvantage is there would likely need to be a library to convert them to 24 bit RGB format for HackTV to display them. Or homebrew your own interpreter.

ljones0, 2020-07-19:
Would a testpattern like this help?

https://github.com/nowster/etp1

03stevensmi, 2022-12-17:
@gregoryfenton I would love to see this please.
This is amazing.

I had a look at your repository's and I can't see or find any fork of hacktv by yourself.
Would it be ok with you if you could share your fork please?
It will be enjoyed and well used.

Cheers,
M.S