fsphil/hacktv #71: Picture problem

Jovan8, 2020-10-16:
When executing the hacktv command in linux terminal everything works for a bit, then a after a while the picture starts shaking, the sound becomes distorted and the console is spamming UUUUUUUU. Ive tried the hacktv exe in windows, the picture shake is smaller, but is constant, unlike in linux where it works perfetly for the first few minutes. Idk how to solve this, i have no experience in linux or programming. Hoping someone can help.

fsphil, 2020-10-16:
This sounds like the CPU is not keeping up. What type of machine/CPU are you using?

Jovan8, 2020-10-16:
> 
> 
> This sounds like the CPU is not keeping up. What type of machine/CPU are you using?

Yea i think that might be it. Im running hacktv on a dual xeon E5335 cpus server. Each one is 4 cores at 2GHz, probably the cpu clock isnt fast enough. Is there any way to optimise hacktv? If not ig i gotta get a better machine. Thanks for the quick reply :smiley:

fsphil, 2020-10-16:
> Is there any way to optimise hacktv?

Don't use the `--filter` option, use the lowest sample rate you can get away with, and if you're transmitting a mode with NICAM audio enabled use `--nonicam` to disable it.

The current version of hacktv does not make good use of multi core CPUs. I have been playing around with a multi threaded version that shows promise, but it may be some time before that becomes available.

Jovan8, 2020-10-16:
I lowered the sample rate to 15 and now it works perfectly on windows! Thanks a lot :smile:

closed 2022-02-07 by fsphil