fsphil/hacktv #76: Add version number for hacktv

Potomac, 2021-05-02:
Currently hacktv doesn't have version number (no cli option --version or -v), it would be interesting to have this feature, 

a scheme like :
major.minor[.build[.revision]] (example: 1.2.12.102)

fsphil, 2021-05-03:
There is no version number, but I've played about with using a yyyymmdd-commit style version:

```
$ ./hacktv --version
hacktv 20210116-751e496
```

Though I'm not completely happy with it.

alexmyczko, 2021-09-21:
this is related to https://github.com/fsphil/hacktv/issues/38

fsphil, 2024-07-21:
In the absence of any better ideas, I've gone with the hacktv date-sommit format:
```
$ ./hacktv --version
hacktv 20240721-e3cfc6f
```

I may change how this works in future but the version format will always at least be hacktv yyyymmdd

closed 2024-07-21 by fsphil