fsphil/hacktv #79: a2stereo option not automatically disables nicam

itewreed, 2021-06-30:
To make a2stereo work, it is also necessary to use the nonicam parameter on the commandline. E.g --a2stereo --nonicam
Not doing so seems to have both, a2stereo and nicam, active. In this case a receiving device cannot detect a2stereo and the transmission still seems mono.
Test was done on two VCRs with stereo reception capability.

I suppose, that by using a2stereo option, the nicam should automatically be deactivated.

fsphil, 2021-07-06:
Agreed. `--a2stereo` will disable NICAM now.

closed 2021-07-06 by fsphil

itewreed, 2021-08-18:
As with the latest pull of August 13th, the issue still is present. To make a2stereo work I still need to use the --nonicam parameter.

fsphil, 2021-08-19:
It seems fine here. Can you share the command line you are using?

itewreed, 2021-08-19:
Found the mistake. I forgot to run the make command after pulling the update. Sorry for the inconvenience.

commit 73b55287 (2021-07-06): Disable NICAM when A2 stereo is enabled. Fixes #79