fsphil/hacktv #82: Add color field identification for SECAM

Potomac, 2021-10-02:
Synchronization of SECAM color transmissions can be done in 2 ways, according to the website "World Analogue Television Standards and Waveforms" : 

- Line identification: by chrominance subcarrier reference signals inserted into the back porch
of the line blanking period
- Field identification: by chrominance subcarrier reference signals occupying nine lines of field-
blanking period on Lines 7 to 15 in Fields 1 & 3 and Lines 320 to 328 in Fields 2 & 4

"Line identification" is used since december 1979 in France, but before december 1979 "field identification" was used, it would be great if hackTV can have "field identification" option for secam, such signal can be inserted in the VBI zone, and both methods can be transmit simultaneously.

Having "field identification" option for hackTV will help owners of old Secam TV built in the 60's and 70's (France, and probably Russia), these old secam TV set can display color only if field identification is present on VBI lines (9 "bottles"), 

currently the only way to display color on these old secam TV set is to use an pal to secam transcoder with field identification recreation, these kind of device are not easy to find on ebay.

I provide as attached file documentation about field identification method (a PDF file, see secam chapter
[World Analogue Television Standards and Waveforms.pdf](https://github.com/fsphil/hacktv/files/7272326/World.Analogue.Television.Standards.and.Waveforms.pdf)
).

fsphil, 2023-01-21:
I've added an option `--secam-field-id` that attempts to do this. I have no hardware that can make use of this so it's completely untested.

Potomac, 2023-01-23:
Great news :+1: 

Thanks.

closed 2023-11-07 by fsphil