fsphil/hacktv #89: Multiple Noob Questions

RandomPSPguy, 2022-05-29:
So I don't have a hackrf yet but am planning on getting one for hacktv. I have no experience with sdr's.
Would this work well with a raspberry pi zero w? I believe the ffmpeg would be too much for live encoding but I saw you are able to render to a file.
example:
-o B:\output.bin -m ntsc-fm -s 18000000 "O:\SHOWS\05 Warren.m4v"

Would I be able to make a 15 hour broadcast and render it to that file and play that back on the hackrf without lagging out on the pi zero? If so how would I go about doing that?
Reason I want the pi zero so bad is I am going mobile and want as little power consumption as possible.

How is the osmo-fl2k? It is tempting since it's cheap but I heard it drifts frequency. Also would I be able to use the osmo with this on usb 2.0?

Unrelated to the software itself but where would I find an amplifier for low vhf? specifically channel 2 ntsc. I'd like somewhere between 5 to 50 watts. Yes I know the laws regarding this.

fsphil, 2022-05-29:
If you're are intending to transmit regular NTSC video you'd be better skipping the hackrf and using omxplayer and the Pi's own composite NTSC output into an RF modulator.

RandomPSPguy, 2022-05-30:
Thanks, I was planning on doing that till I saw this project, I guess i'll go back to a modulator though. 
My thinking was just that rca and a modulator is inherently a little messy so by using this I could skip all of that and get the cleanest signal possible.

fsphil, 2022-05-30:
The hackrf adds some noise to the output and this is visible in the image. A good quality RF modulator will probably beat it for image quality.