fsphil/hacktv #92: How do I transmit pal from fl2k VGA Device?

03stevensmi, 2022-08-11:
Hi again,
I need some info here if possible since Google is well... Not been helpful.

I Have a fl2k vga device with a vga to bnc converter.

How exactly do I get this to transmit to my pal tv?
There is no documents I can find anywhere explaining what to do.

From what I kind of picked on is the fl2k doesn't have a modulator.

So my question is do I have to hook up the red wire from the vga2bnc to a modulator
(such as a vcr etc) with a booster to succseffully get a picture to show up?

If someone here can shed some light on what I have to do along with an example,
That would be very helpful. 
Any help with this will be appreteated.

Thank you

captainjack64, 2022-08-11:
The "red" wire (from the RGB pin) carries composite video, so just plug that directly into the composite-in socket on your TV - or, like you say, an RF modulator/VCR.

Use option -m pal (instead of -m i) on the command line.

Note that there's no audio, just video and it will be a little dim due to fl2k limitations.

03stevensmi, 2022-08-11:
Thanks man. Any way to get audio? Or is the only other option is to use a septate device with slpitter?
Also when I transmit, I only get White static. Not snow. I have used zdig usb on composite and not interface.
Should I have used interface 0 instead?

captainjack64, 2022-08-11:
No audio at the moment - unless you use a MAC format, in which case you'll get audio through the decoder.

Not sure about other questions.

03stevensmi, 2022-08-11:
Thanks man. Ill make it a bit more clear. When installing winusb via zadig, should i have chosen interface0?
Or should I have chosen composite when installing libusb on the fl2k vga? Just need to know which one I was supposed to use.

captainjack64, 2022-08-11:
Not a clue - I don't touch Windows!

03stevensmi, 2022-08-11:
Ok. The only other device I have wichnisnt windows is a raspberry pi 4 4gb.
But from what I read, that isn't powerful enough to sucaesfully work am I right?

I haven't tried virtualbox or VMware, but from what I have been reading, they don't work ether.

One more question,
Is it normal for a USB to vga adapter to have a loose wire that isn't connected to anything?
I have had a look at the device and all the wires are all connected. Is it a spare wire that didn't do anything?

03stevensmi, 2022-08-11:
Doesn't matter! I got a picture! For all your help, I have got to say thank you so much cj!
The problem I made was I was hoocking it up from bnc bia rf. I just quickly made a bnc to race and BOOM!!!
Jk, i got a c,ear picture. Using a Good booster, I really don't know what you meant by dark picture.

what I received was very clear. But if I had to be honest, I belive thats because the modulator I am using has a brightness control. So all is Good. Thanks for everything cj. Serioulsy you and everyone here involved in hacktv are doing the future some dam Good let me tell ya! Cheers man.

closed 2022-08-12 by 03stevensmi