fsphil/hacktv #93: Is it possible to pipe the audio from hacktv to the pc's audio jack?

03stevensmi, 2022-08-11:
Hey guys.
I'm loving the use of hacktv on the vga fl2k.
But as you all know thees no audio.

So my question is,
Would it be possible for hacktv to have the option to output the audio sepratly to the pc's audio jack?

That way when hacktv is playing the video,
Red and white composite cables can be used sepratly with the use of the audio coming from the pc,
Instead of the vga.

Would this be possible?
If so, how?

Cheers

fsphil, 2022-08-12:
Not at the moment. It should be relatively simple to add though, but there may be issues with A/V sync.

03stevensmi, 2022-08-12:
> Not at the moment. It should be relatively simple to add though, but there may be issues with A/V sync.

Ah you beet me to it. Was just updating the question.
Is there any way I could test?
I'm also aware about the a/v sync issue that could arise, but its better than nothing.
Would a virtual sound device like obs be possible to use?

03stevensmi, 2022-08-16:
Can I test this out please? 

Thanks man.

fsphil, 2022-08-16:
There's nothing to test yet, it still needs to be coded :-)

03stevensmi, 2022-08-20:
> There's nothing to test yet, it still needs to be coded :-)

will it be possible to code this?

fsphil, 2022-08-20:
Yes, but I can't say when I'll get time. There are a few other things I need to fix/improve first.

03stevensmi, 2022-08-21:
> Yes, but I can't say when I'll get time. There are a few other things I need to fix/improve first.

Ok. Thanks man. Will keep an eye out on here for updates and fixes etc.
Cheers man. Really appreciate all the hard work that you all do.

03stevensmi, 2023-01-31:
I will leave a quick mention of this on the discord also,
but i have found a way (or github fork)
that will enable pcm (raw wav) audio output through the green (or optional blue) pin
along with composite baseband video on ther fl2k.

Git:
https://github.com/vrunk11/fl2k_2

audio encoder:
pcm_s16le

sox:
sox.exe -b 16 -c 1 -r 88.2k -e signed-integer -t raw ".\Video-audio-input.wav" -b 8 -r 7777777 -e signed -t s8 - | fl2k_file.exe -s 7777777 audio.pcm

Fl2k_2 video (with audio):
hacktv -o - -m pal -t uint8 -s 13500000 --wss 4:3 --logo multichoice --interlace --vitc test:pm5544 | fl2k_file2 -s 13500000 -signR 0  -R - -R8 -G audio.pcm

would this be a way to get audio for the fl2k device?
i have tried it and can confirm that it does accually work with hacktv vid&audio.
only problem is i have to use ether sox or ffmpeg to convert or resample the audio from video file to pcm_s16le.

if there is a way to do the video file's audio conversion to pcm in real time ether by pipe or in hacktv itself,
could that be supported in hacktv?