fsphil/hacktv #94: What do I need to use my fl2k with a pace prima receiver/decoder?

03stevensmi, 2022-08-16:
Hey guys.
I have purchase a pace prima analog satellite refugee with built in decoder.
The only inputs are coaxal (obviously for sattelite) and rf in/out.

The fl2k can only output composite.

What do I need to do to be able to use my fl2k with my pace prima box?

Can I input a rf signal from a vcr/modulator to the prima?

Or do I have to purchase a bnc or rca to coaxal?

Amy help on this would be helpful

Cheers

fsphil, 2022-08-16:
You'd need an FM modulator that can output at the satellite IF frequencies (around 1 GHz). Or skip the fl2k and use an SDR like a hackrf to generate the FM signal directly.

03stevensmi, 2022-08-16:
Dam! Thats what I was fearful of. Ah well, thanks anyway

03stevensmi, 2022-08-21:
> 

Just had a look at the support for soapysdr.
will the rtl-sdr work?

I really don't have the money for a hackrf and couldn't find any info on a 
analog if satellite modulator on ebay. 

It needs to be around the £40-£50 mark.

Can the rtl-sdr can do satelitte transmition or even basic pal/secam signals?
Thanks

fsphil, 2022-08-21:
Afraid not, the rtl-sdr is a receiver only. The hackrf one is still the best SDR option.

For modulation look for 23cm ATV modulators. The 23cm amateur radio band uses frequencies that a standard satellite receiver can handle.

03stevensmi, 2022-08-21:
> Afraid not, the rtl-sdr is a receiver only. The hackrf one is still the best SDR option.
> 
> For modulation look for 23cm ATV modulators. The 23cm amateur radio band uses frequencies that a standard satellite receiver can handle.

Whats the average price for one of those ATV Modulators?
It needs to be something around £45-£50.

If those modulators are too expensive,
Then is there any sdr that hacktv supports that is around £50?

fsphil, 2022-08-21:
> Then is there any sdr that hacktv supports that is around £50?

I really wish there was, I'd definitely get one and add support for it.
Right know the hackrf one is the cheapest option.

Unknown on the ATV modulator.

03stevensmi, 2022-08-22:
> > Then is there any sdr that hacktv supports that is around £50?
> 
> I really wish there was, I'd definitely get one and add support for it. Right know the hackrf one is the cheapest option.
> 
> Unknown on the ATV modulator.

Ok, thanks anyway man. I'll keep an eye out.
Thanks for all your help fsphill!
I'll also keep an eye out on the audio support through pipe so I've 
still got something Good Out of this project.

Cheers man.
03stevensmi
;)

closed 2022-08-22 by 03stevensmi