fsphil/hacktv #98: Software-Defined TBC Support

03stevensmi, 2023-01-30:
@philpem @captainjack64 
Would it be possible to have hacktv's baseband signals support
TBC (Time Base Corrector)
Like the ones used in CVBS
(via the ld-chroma-encoder)
- https://github.com/oyvindln/vhs-decode/wiki/CVBS-Composite-Decode
And [vhs-decode](https://github.com/oyvindln/vhs-decode/) / [ld-decode](https://github.com/happycube/ld-decode/)

Here is some info:
https://www.digitalfaq.com/forum/video-restore/2251-tbc-time-base.html#:~:text=A%20%27timebase%20corrector%27%20or%20%27time%20base%20corrector%27%20--,then%20it%20is%20corrected%20before%20being%20output%20again.

Is it possible for to this be done?
;)

fsphil, 2023-01-30:
I'm not sure I understand the request, TBC wouldn't do much for hacktv's output as there shouldn't be anything to correct, there's no variability like you'd get from a mechanical video recorder.

03stevensmi, 2023-01-31:
in this fork:
https://github.com/vrunk11/fl2k_2

there is the option for support with .tbc files wich is used/generated with programs such as this:
https://github.com/happycube/ld-decode/wiki/ld-chroma-encoder

a tbc file (wich would make hacktv great use with many analog equipmemt)
is explained with info here:
https://en.wikipedia.org/wiki/Draft:.TBC_Time_Base_Corrected_Format

woud it be possible if hacktv could generate in real time and output a "tbc" file (simular to those 2 github links above) 
for use with people like myself who cant afford a tbc player.

If it isnt possible to generate a tbc in real time,
could you please support the tbc file format with use on hacktv?

Hope this has helped explain it better.
Cheers,
03stevensmi

;)

pepie34, 2023-02-23:
There is a quiproquo.

TBC file format has nothing todo  with actual Time Base Correction for VCR.
It is just a file format to store a RF capture from LD/VCR/Broadcast + metadata for the xxx-decode tools, where xxx is ld or vhs or cvbs projects.

It will be indeed be usefull if hacktv and xxx-decode projects can share the same file format.

03stevensmi, 2023-02-24:
> There is a quiproquo.
>
> TBC file format has nothing todo with actual Time Base Correction for VCR. It is just a file format to store a RF capture from LD/VCR/Broadcast + metadata for the xxx-decode [...]

I agree, it would be very useful if hacktv did share the same file format, 
my use would be for my 80's video mixer.
Both composite inputs need a tbc for it to be used otherwise it just isn't usable.
Having hacktv output a tbc from one (or simultaneously 2) of my video files 
would make use of old video mixers and old broadcast equipment.

We will just have to see what happened as Phil is extreamly busy with these projects & plans.
But I guarantee... If It would be anyone that could make something like this Happen,
It would be people like him and captain jack that could perform these type of miracles.

;)

fsphil, 2023-02-27:
> It will be indeed be usefull if hacktv and xxx-decode projects can share the same file format.

What use would that have other than being able to load a hacktv generated signal into cvbs-decode?

pepie34, 2023-02-27:
It is flac compress so less space used

philpem, 2023-02-27:
As fsphil asked -- what is the use case for this? What will this allow you to do which you can't already do?

Loading the output of HackTV into ld-decode seems silly: you're just going to get the same video data back, only with added distortions from HackRF's scaling and encoding, and ld-decode's PAL (or whatever standard you're using) colour transform decoder.

If you want to play out a .TBC file you should be able to convert it to, say, raw YUV or RGB, Huffyuv or Dirac, which are all lossless and should be supported by the existing FFMPEG front-end.

pepie34, 2023-02-27:
Because we want to do Machine Learning on the xxx-decode side and we want to feed the learning step with a control/supervised dataset generated with the help of hacktv

philpem, 2023-02-27:
You could write raw samples to a file using `-o filename` and one of the `real` formats. That'd give you a CVBS baseband waveform which you could compress to FLAC if needed.

If you want to compress to FLAC on the fly, you could feed the output into a named pipe (`man mkfifo`) and have FLAC read the samples from there and output to a FLAC file.

closed 2023-11-07 by fsphil