fsphil/hacktv #102: Is support for Raspberry Pi possible?

DemonKingFukai, 2023-05-22:
RPItx, PiFM, and various other softwares and hardwares use pin 7 on a raspberry pi to transmit various signals. Is it possible to utilize this for transmission using HackTV? There is support for FL2k so I thought Raspberry Pi would be supported as well.

fsphil, 2023-05-22:
Technically yes, but only the two narrowband TV modes `30-am` and `nbtv-am` would work. The maximum bandwidth of 250kHz just isn't enough to fit a standard 525/625-line TV signal - it would need to be at least 25 times wider.

03stevensmi, 2023-05-23:
Wouldn't something like one of those rpi tx hat work? There are a few that can boost the rpi to a much wider band and frequency.

fsphil, 2023-05-23:
Have you got a link to one?

03stevensmi, 2023-05-24:
> Have you got a link to one?

Sure,
Here is one
https://github.com/cariboulabs/cariboulite

fsphil, 2023-05-24:
Its analogue bandwidth is 2.5 MHz, better but still too narrow for an NTSC/PAL TV mode.
You may be able to get monochrome video from it but it wouldn't look all that good.

closed 2023-11-07 by fsphil