fsphil/hacktv #103: TV Guides/Listings/Channel Information

thatguycy, 2023-06-01:
Is there any way to transmit Channel Information or Guides over Analogue using HackRF? I didn't see any other issues about this and I was curious so I made one.

fsphil, 2023-06-01:
Do you mean Guide Plus (https://en.wikipedia.org/wiki/Guide_Plus)? Or something else?

thatguycy, 2023-06-02:
Like the TV Guides found actually on the TV

[image]
[image]

fsphil, 2023-06-02:
Those are all DVB digital TV services. The only analogue TV equivalent I ever saw for that was simply the TV guide pages on teletext, which you can definitely do with hacktv - you just have to write the pages.

thatguycy, 2023-06-02:
I don't know anything about teletext, but thank you so much! I'll research it!

fsphil, 2023-06-02:
No worries. There used to be dedicated EPG-style services like Guide Plus above in the US and other countries for analogue TV, but I don't have any real technical info about how they worked.

closed 2023-06-02 by fsphil