fsphil/hacktv #104: Would a satellite transmission work with the sdr? hackrf for example

j1m1l0k0, 2023-06-05:
Is there any way to adapt the code for DVB-S2/S/T demodulation and sending via satellite?

fsphil, 2023-06-06:
Do you mean can you use a DVB channel as an input for hacktv? If so then yes, using software like tvheadend or mumudvb - then pass the URL for your channel as an input to hacktv.

j1m1l0k0, 2023-06-06:
It actually wasn't that.  I would like to know if there is a way to implement in source code demod dvb-s2, dvb-s, signals via satellite equal to dvb-s2 demod gui.  See this app https://forum.amsat-dl.org/index.php?thread/101-software-dvb-s-demodulator/

fsphil, 2023-06-06:
No, that's outside the scope of the project. However the TS output from that program could be used as a source for hacktv.

j1m1l0k0, 2023-06-06:
Thanks Phil ;)