fsphil/hacktv #105: HackRF output only a single, small carrier

kylemikableh, 2023-08-18:
Hey so I recently got a HackRF and I have tried multiple computers, versions, builds, and yet I get the same signal no matter what. Any mode, any source its always a single little carrier exactly on the center frequency configured (so at 910MHz when watching using another SDR and sdr++ all I see is a small line on the waterfall on that frequency). Any idea what is the issue? I don't see what I can be doing wrong, using gnu radio I can receive and transmit FM for example completely fine. I have tried different video sources and the sample bars. No logs seem to say anything is wrong. Thank you for your time, this has been driving me crazy.

fsphil, 2023-08-18:
What command line arguments are you using?

kylemikableh, 2023-08-18:
Hi, thanks for responding.
 `-m m -f 561250000 -s 13500000 -g 0 test:colourbars`
This is the most basic one I have used, I have varied symbol rate (14000000 for example), passed in some mp4's from archive.org, and other modes just to see if anything seems to happen when viewing from my other SDR. I have the newest revision of the HackRF, not sure if that would cause any issues but this same issue seems to happen on Windows and Ubuntu binaries (HackRF works with gnuradio for Tx/Rx FM fine as previously stated). Also, this is an example frequency, I have tried many channels. I am located in the US, so NTSC-M is the desired mode.

fsphil, 2023-08-18:
`-g 0` sets the hackrf output to its lowest setting, try increasing that.

kylemikableh, 2023-08-18:
With `-g 15` I get more, but it just looks more like harmonics: [image]

fsphil, 2023-08-18:
That looks like a normal TV signal, just a little weak. The `-g` setting goes up to 47 so you're still on the low side.

kylemikableh, 2023-08-18:
I am now embarrassed. It worked! Thank you so much, I completely spaced that 47 is the max gain. Thank you for making this tool and helping me.

closed 2023-08-18 by kylemikableh