fsphil/hacktv #106: Cannot compile  hackrf_get_transfer_buffer_size

zonemikel, 2023-08-27:
Where does this hackrf_get_transfer_buffer_size live ? Can I just replace it with a hardcoded number if I know which hackrf I have? I have an older version of mint so I cant 'apt-get install'. Thanks for all the work, looks great, hope I dont have to wipe the box to use it. 

`michael@michael-l2:~/hacktv/src$ sudo apt-get install libhackrf-dev libavutil-dev libavdevice-dev libswresample-dev libswscale-dev libavformat-dev libavcodec-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libhackrf-dev is already the newest version (2018.01.1-2).
libavcodec-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavdevice-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavformat-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavutil-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libswresample-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libswscale-dev is already the newest version (7:3.4.11-0ubuntu0.1).
The following packages were automatically installed and are no longer required:
  libllvm9 libllvm9:i386 libwine libwine-development libwine-development:i386 libxcb-xtest0 wine32-development:i386
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 751 not upgraded.
michael@michael-l2:~/hacktv/src$ make
gcc -o hacktv hacktv.o common.o fir.o vbidata.o teletext.o wss.o video.o mac.o dance.o eurocrypt.o videocrypt.o videocrypts.o syster.o acp.o vits.o vitc.o nicam728.o av_test.o av_ffmpeg.o rf_file.o rf_hackrf.o -g -lm -pthread  -lavcodec -lavformat -lavdevice -lswscale -lswresample -lavutil -lhackrf
rf_hackrf.o: In function `rf_hackrf_open':
/home/michael/hacktv/src/rf_hackrf.c:371: undefined reference to `hackrf_get_transfer_buffer_size'
/home/michael/hacktv/src/rf_hackrf.c:373: undefined reference to `hackrf_get_transfer_buffer_size'
collect2: error: ld returned 1 exit status
Makefile:29: recipe for target 'hacktv' failed
make: *** [hacktv] Error 1
michael@michael-l2:~/hacktv/src$ grep -r hackrf_get_transfer_buffer_size
Binary file rf_hackrf.o matches
rf_hackrf.c:    r = s->vid.sample_rate * 2 * 4 / 10 / hackrf_get_transfer_buffer_size(rf->d);
rf_hackrf.c:    _buffer_init(&rf->buffers, r, hackrf_get_transfer_buffer_size(rf->d));
`

zonemikel, 2023-08-27:
Better code snip
`
michael@michael-l2:~/hacktv/src$ sudo apt-get install libhackrf-dev libavutil-dev libavdevice-dev libswresample-dev libswscale-dev libavformat-dev libavcodec-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libhackrf-dev is already the newest version (2018.01.1-2).
libavcodec-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavdevice-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavformat-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libavutil-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libswresample-dev is already the newest version (7:3.4.11-0ubuntu0.1).
libswscale-dev is already the newest version (7:3.4.11-0ubuntu0.1).
The following packages were automatically installed and are no longer required:
  libllvm9 libllvm9:i386 libwine libwine-development libwine-development:i386 libxcb-xtest0 wine32-development:i386
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 751 not upgraded.

michael@michael-l2:~/hacktv/src$ make
gcc -o hacktv hacktv.o common.o fir.o vbidata.o teletext.o wss.o video.o mac.o dance.o eurocrypt.o videocrypt.o videocrypts.o syster.o acp.o vits.o vitc.o nicam728.o av_test.o av_ffmpeg.o rf_file.o rf_hackrf.o -g -lm -pthread  -lavcodec -lavformat -lavdevice -lswscale -lswresample -lavutil -lhackrf
rf_hackrf.o: In function `rf_hackrf_open':
/home/michael/hacktv/src/rf_hackrf.c:371: undefined reference to `hackrf_get_transfer_buffer_size'
/home/michael/hacktv/src/rf_hackrf.c:373: undefined reference to `hackrf_get_transfer_buffer_size'
collect2: error: ld returned 1 exit status
Makefile:29: recipe for target 'hacktv' failed
make: *** [hacktv] Error 1

michael@michael-l2:~/hacktv/src$ grep -r hackrf_get_transfer_buffer_size
Binary file rf_hackrf.o matches
rf_hackrf.c:    r = s->vid.sample_rate * 2 * 4 / 10 / hackrf_get_transfer_buffer_size(rf->d);
rf_hackrf.c:    _buffer_init(&rf->buffers, r, hackrf_get_transfer_buffer_size(rf->d));

`

fsphil, 2023-08-27:
In the current version of libhackrf it always returns 262144:

```
#define TRANSFER_BUFFER_SIZE  262144

[...]

/**
 * Get USB transfer buffer size.
 * @return size in bytes
 */     
size_t ADDCALL hackrf_get_transfer_buffer_size(hackrf_device* device)
{
        (void) device;
        return TRANSFER_BUFFER_SIZE;
}
```

closed 2023-11-07 by fsphil

777arc, 2024-03-30:
Still getting this same error, on Ubuntu 22 with the latest master

```
rf_hackrf.c: In function ‘rf_hackrf_open’:
rf_hackrf.c:365:40: warning: implicit declaration of function ‘hackrf_get_transfer_buffer_size’ [-Wimplicit-function-declaration]
  365 |         r = sample_rate * 2 * 4 / 10 / hackrf_get_transfer_buffer_size(rf->d);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -o hacktv hacktv.o common.o fir.o vbidata.o teletext.o wss.o video.o mac.o dance.o eurocrypt.o videocrypt.o videocrypts.o syster.o acp.o vits.o vitc.o nicam728.o av.o av_test.o av_ffmpeg.o rf.o rf_file.o rf_hackrf.o -g -lm -pthread  -lavcodec -lavformat -lavdevice -lswscale -lswresample -lavutil -lhackrf 
/usr/bin/ld: rf_hackrf.o: in function `rf_hackrf_open':
/home/marc/hacktv/src/rf_hackrf.c:365: undefined reference to `hackrf_get_transfer_buffer_size'
/usr/bin/ld: /home/marc/hacktv/src/rf_hackrf.c:367: undefined reference to `hackrf_get_transfer_buffer_size'
collect2: error: ld returned 1 exit status
make: *** [Makefile:36: hacktv] Error 1
```