fsphil/hacktv #111: -ffmt syntax problem

michaspie8, 2024-04-23:
##### \*If I'm talking nonsense, then I apologize, but I'm not that proficient in these matters.\*
The `--ffopts` option requires writing subsequent options and values in the form of [option1]=[value1]: [option2]=[value2], etc... 

The problem is that (I may be wrong, I don't know how to check) I have the impression that when I choose options like
`c:v=h264_nvenc`, the program thinks rather about
`-c -v h264_nvenc` than about 
`-c:v h264_nvenc`. 
Similarly with '=', if a value contains an equal sign, it might not work correctly.

How should I deal with this? Does the code anticipate such a situation? 
Should I write these commands, for example, like `"c:v"=h264_nvenc` or `c\:v=h264_nvenc`?

I'm sorry I didn't look at the code, but I'm at a level of understanding code where I could actually figure it out, but it would take a lot of time, and if I found a problem and fixed it, I definitely wouldn't compile it myself just to see how it works, because I only have Windows (I tried and installing libraries was a torment for me and I couldn't get out of it), alternatively I would use Tails Linux from a USB, although it would still be quite difficult. Thanks in advance for your help!

michaspie8, 2024-04-23:
Alright, I got it, it worked when I used \

fsphil, 2024-04-23:
Can you describe what you're trying to do?

03stevensmi, 2024-05-16:
> Can you describe what you're trying to do?

Something quite important for me but for a different reason to answer your question Phil,
What he's trying to do Phil, is He's trying to encode a file with --ffopts using hacktv instead of just having the option to choice format "--ffmt" (I'll get I to how that would be important to me in a bit.)

From looking at his examples, it looks like he wants to use hacktv to play his videos with nvenc (gpu encoding).

So basically, he is asking if a command like "--ffopts "c:v=h264_nvenc" would work or not.
And if it does work, can "c:v=h264_nvenc" be used, or would it need to be something like 
"--ffopts "-c:v h264_nvenc".

For me, this would be helpful with my analogue capture card!
I have to mess around with piping with ffmpeg and it can be a bit fiddley sometimes.

If I use --ffmt v4l2 with --ffopts "framerate=25" I get no sound. So I have to combine the video and audio from v4l2 with pulseaudio to mkv, avi or mpegts by piping and messing around just to get sound from my capture card to hacktv.

So for me, I would honestly too would like to know if hacktv's --ffopts feature would support or work with 
Something like `hacktv -o fl2k -m pal-fm --ffmt mpegts --ffopts "-f v4l2 -framerate 25 -vcodec h264_nvenc -acodec aac" ffmpeg:"[VIDEO FROM v4l2] -i [AUDIO FROM pulseaudio]`

fsphil, 2024-05-16:
That'll be a no then. `--ffopts` is used to pass options to the format selected with `--ffmt`. You can see an example of some here: https://ffmpeg.org/ffmpeg-formats.html