fsphil/hacktv #112: Question about ATSC and I'm probably asking in the wrong place

naikrovek, 2024-04-25:
I want something like this for ATSC, but which still sends old SD video on subchannels of an ATSC channel.

... I understand that ATSC is not analog and that this software does analog, but does anyone know of software like what I'm describing?  GNURadio is python and doing all the conversion to transport stream and encoding to ATSC is pretty slow in Python, so I was hoping there was something like this excellent tool for SD video over ATSC.  If anyone knows of software like that for the HackRF, please let me know.

I figure the people working on this probably have a better idea of what's going on in this realm than I do, and Google is giving me nothing.

Thanks, and sorry for the off-topic interruption.

fsphil, 2024-04-27:
I'd love something like this to exist, but I don't know of anything that fits. For me it would be DVB but I'd expect any such program would be able to do most/any of the common digital modes.

naikrovek, 2024-05-02:
I did eventually find this, though I have yet to try it out: https://github.com/philburr/atsc

closed 2024-07-13 by fsphil