fsphil/hacktv #114: Linux mint "make" is giving me errors

t0nitogp, 2024-07-03:
I'm trying to build hackTV and I ran into an issue, I installed the requirements 

apt-get install libhackrf-dev libavutil-dev libavdevice-dev libswresample-dev libswscale-dev libavformat-dev libavcodec-dev

When I go to "make" command it starts running and stops at:

/usr/bin/ld: rf_hack.o: in function 'rf_hackrf_open':
/home/mint/hacktv/src/rf_hackrf.c:365: undefined reference to 'hackrf_get_transfer_buffer_size'
/usr/bin/ld: /home/mint/hacktv/src/rf_hackrf.c:367: undefined reference to 'hackrf_get_transfer_buffer_size'
collect2: error: ld returned 1 exit status
make: *** [Makefile:36: hacktv] Error 1

t0nitogp, 2024-07-03:
[image]

fsphil, 2024-07-03:
This is a duplicate of #106

Basically your system is still using a very old version of libhackrf that lacks the `hackrf_get_transfer_buffer_size()` function.
You can simply replace that function call with `262144`.

t0nitogp, 2024-07-03:
You mean edit the rf_hackrf.c and replace on both lines 'hackrf_get_transfer_buffer_size()' with 262144?

I'm sorry if this is a stupid question but I'm very new to Linux, I'm using the latest version of Mint, is there a way to maybe update a newer version of libhackrf and try again?

t0nitogp, 2024-07-04:
> This is a duplicate of #106
> 
> Basically your system is still using a very old version of libhackrf that lacks the `hackrf_get_transfer_buffer_size()` function. You can simply replace that function call with `262144`.
> 
> I might just remove this call until Ubuntu and Mint fix their packages.

I tried replacing `hackrf_get_transfer_buffer_size()` with 262144 and just got more errors. I know nothing about code...

fsphil, 2024-07-04:
I've pushed a change that removes that function call, that should compile for you.

fsphil, 2024-07-04:
Did it work?

t0nitogp, 2024-07-04:
I didn't have a chance to it try yet, but as soon as I do I'll let you know :)

fsphil, 2024-07-04:
Thanks for reporting back!

closed 2024-07-04 by fsphil