fsphil/hacktv #115: Vertical stripes during signal transmission at -s 16000000

nerevist, 2024-07-17:
I transmit an analog signal using the HackTV build for Windows.
Both the fsphil assembly and the Captain Jack assembly have vertical stripes.
[image]
[image]

fsphil, 2024-07-17:
This seems to be an issue with recent revisions of the hackrf hardware.
Can you run `hackrf_info` and copy the output here?

nerevist, 2024-07-17:
> Can you run `hackrf_info` and copy the output here?
I'm sitting under Windows and this utility is not available for windows

nerevist, 2024-07-17:
[image]

fsphil, 2024-07-17:
Huh, I was sure `hackrf_info` showed the board revision, but I don't see it there.
Anyway, the hackrf r9 is known to have those lines. Try using different sample rates, such at 15 or 17 MHz.

nerevist, 2024-07-18:
> Try using different sample rates, such at 15 or 17 MHz.

On 15 MHz all normal, but on 17 MHz picture and sound bugged

fsphil, 2025-04-15:
Closing as this is not a hacktv issue. Sadly it doesn't look like there's going to be a fix for hackrf-r9.

closed 2025-04-15 by fsphil