fsphil/hacktv #117: ntsc signal spikes

DrSDR, 2024-09-17:
not sure if it's the hacktv-gui or the hacktv software,  but when making ntsc signals, i save signal as float file,  when i play it back in gnuradio,  i see spikes in the i and q signals.      at the sync pulses, i see it go from 1 to -1 in short blips.      
hard to explain,  i sent the iq file to bladerf and see the spectrum flash ,  not sure if the i or q signal is getting clipped when the file was being generated.         any insights ?       thanks

fsphil, 2024-09-17:
Can you share the hacktv command line that produces the IQ file?

fsphil, 2024-09-18:
I'm not able to reproduce it here, the syncs are not overflowing the output.
Are you doing any signal processing in gnuradio that might be introducing them?

DrSDR, 2024-09-18:
ok thanks,  i am just using the gnuradio file source  and feeding that into a qt gui sink block,   as for hacktv , i just followed the getting started guide for linux,  and it told me to apt get install hacktv  .
perhaps i'll try building hacktv from src  folder and see if issue goes away,  thanks for looking into it.   cheers

fsphil, 2024-09-18:
Ah, yeah that might be an older version. I tweaked the levels of ntsc/m earlier in the year: https://github.com/fsphil/hacktv/commit/40c0aa61516c52d64e1d811ee646aa55760a69d6

DrSDR, 2024-09-19:
also can you provide me the complete steps to build the project?   do i have to do gitclone commands in the terminal before the cd src  and make  ?

fsphil, 2024-09-19:
You can git clone, or download the source in a zip: https://github.com/fsphil/hacktv/archive/refs/heads/master.zip

There's some notes in the README to build it, it's quite simple. Hardest part is installing the dependencies.

DrSDR, 2024-09-23:
i compiled the src folder,  and now the ntsc spikes are gone.
when will the :   apt get install hacktv ,   have the latest src ?            
the hacktv in the dragonos build seems to have the old hacktv as well.
anyway compiling the latest src fixed the issue,  thanks again.

fsphil, 2024-09-24:
Good to hear it's working.
Unknown on the package update, I don't really do releases so it's up to the packager. It would do no harm to send them a message.